Klipper3d/klipper #1909: changing filament mid print (urgent)

tomohulk, 2019-08-24:
Sorry for opening a issue for this, but i'm not really sure where else to ask this and hoping for a quick reply.  I am at the tail end of a 2 day print, i have bout 4 hours left (estimated) and im running very very low on filament.  I have more, but i have never done filament change mid print.  And im not really sure how to.  I have read a few issues here talking about a change filament macro, but i don't need anything fancy, i just want this print to finish successfully.   Can i just use the Octoprint Menu 'Pause' and then move the head off the print and then change the filament, and the use the corasponding 'Resume' action? 
I would normally play with these and figure it out myself, but at the end of such a long print, i really do not want to take any chances.

I am using and Ender3 with Klipper running on Octopi.  I updated Klipper a few days ago if that matters.  I can grab a log, but this more of a "How To" and not an issue.

very much appreciate any help and information!  thanks!

BlackStump, 2019-08-24:
why don't you just feed the new filament in behind the old as it is printing ...you of course need to be at the printer and when the filament comes off the roll butt the new filament to the old until the extruder grabs it.
If you have not tested the pause resume or used before, to many things can go wrong and you end up with a failed print.
I guess that is a mute point now if you only had 4 hours to go but planning before a long print is required ie do you have enough filament and how do I solve that issue if I runout.

tomohulk, 2019-08-25:
i didn't want to do that because of retraction.

closed 2019-08-25 by tomohulk