Klipper3d/klipper #1910: Klipper is reading incorrect thermistor values

bryanboettcher, 2019-08-24:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3537562/klippy.log)

This is a weird one.  Given the _same physical machine_, and a 320 ohm resistor in place of the extruder thermistor:

* RAMBo on Klipper: 231C
* RAMBo on Marlin2: 232C
* Archim2 on Klipper: **200C**
* Archim2 on Marlin2: 232C

I am not concerned with the 1C difference between Klipper and Marlin2 on the RAMBo, but the Archim2 differences are enormous.  It is a sliding scale, too ... the higher the commanded temperature, the worse the difference is  (eg. at 220C commanded, the actual is 283C).  Every thermistor in both Klipper & Marlin2 is set to "EPCOS 100K B57560G104F".

bryanboettcher, 2019-08-24:
I should add this is only software, in some manner.  All 4 thermistor input channels exhibit this behavior.

FHeilmann, 2019-08-26:
I'm not sure how Marlin corrects for this, but it seems like the Archim2 (I looked at the rev2.1 board) board has some additional circuitry in place for the thermistors which might throw the reading of the temperature value off-course.
Here is an example for T1: 
[image]

I haven't used Marlin in a while and don't know whether Marlin corrects for the inline circuitry somewhere, but that would be my guess why your readings differ on this particular board. Klippers table for the EPCOS thermistor assume no additional circuitry.

bryanboettcher, 2019-08-26:
I noticed that too, except the RAMBo has the same circuitry, down to the
resistor values. Klipper gets the temperature correct with RAMBo. Trying to
add values for "inline_resistor" makes the temperature readings worse.

nophead, 2019-08-26:
The series resistors should have negligible effect assuming the capacitor
is a low leakage type. The diodes and input pin should only leak a few uA.

bryanboettcher, 2019-08-26:
> going back to the original post, @bryanboettcher
> <https://github.com/bryanboettcher> you stated that you're putting a 320
> Ohm resistor in place of the thermistor. Thats a massive difference from
> what the thermistor will usually read (100kohm at room temperature). Maybe
> thats the reason for the inaccuracies. as @nophead
> <https://github.com/nophead> pointed out the resistors on board will not
> affect the reading with a regular thermistor (2k vs 100k) but with a 320
> Ohm resistor in place of the thermistor things may look different.

The behavior persists with an actual EPCOS 100k B57560G1104F thermistor in
place of the resistor, I just can't get fractional temperature readings
from the ADC because I'm manually unplugging & plugging the heated
thermistor into another board.  The effect is amplified the higher the
temperature reading is supposed to be, and the correct temperature is
always HIGHER than Klipper reads it as.

My initial test came by heating the hotend to a commanded 220C with
Klipper, with the sensor set to EPCOS 100k B57560G1104F.  Sitting near my
printer, I had a RAMBo running Marlin (hear me out) with the thermistor
harness all wired through and sitting next to the hotend.  I had a script
running M105 on a loop to get temperature readings.  They usually read -15C
since there was no thermistor.  In one quick motion, which took less than 2
seconds, I would unplug the thermistor from Klipper and plug it into the
Marlin harness.  I would expect a reading of 220C or less if the hotend had
cooled, but instead Marlin was reading 283C!

I then thought this was a Marlin / Klipper difference, so I tried it on a
RAMBo running Klipper set to 220C / EPCOS 100k, moving to the same RAMBo
from earlier running Marlin.  The commanded 220C came out like 219C on
Marlin, which I attribute to 2 seconds of hotend cooling from the safety
shutdown.

I then thought this was a RAMBo / Archim2 difference, so I tried Marlin on
the Archim2 moving to my same reference RAMBo running Marlin.  The
commanded 220C was still right about 220C.

This problem is NOT the thermistor, it is NOT Klipper on the whole, and it
is NOT the Archim2 on the whole.  It only happens with an Archim2 running
Klipper.

I am happy to get my test setup to a Klipper developer for further testing
if needed.

nophead, 2019-08-26:
> but with a 320 Ohm resistor in place of the thermistor things may look different.

The series resistors shouldn't make much difference at any temperature or thermistor resistance because very little current flows through them, so there is very little potential difference across them. All the current flows through the 4K7 and the thermistor. 

320 Ohm is the value the thermistor should have at ~230C, so it should read correctly.

bryanboettcher, 2019-08-26:
> I see, apologies for the confusion. Frankly I'm now out of ideas what the
> problem could be here.

Well, how should I proceed?

FHeilmann, 2019-08-26:
going back to the original post, @bryanboettcher you stated that you're putting a 320 Ohm resistor in place of the thermistor. Thats a massive difference from what the thermistor will usually read (100kohm at room temperature). Maybe thats the reason for the inaccuracies. as @nophead pointed out the resistors on board will not affect the reading with a regular thermistor (2k vs 100k) but with a 320 Ohm resistor in place of the thermistor things may look different.

bryanboettcher, 2019-08-26:
> If you have a digital multimeter you could measure the voltage on the
> thermistor input and confirm it is correct for a 320 Ohm resistor in a
> potential divider with 4K7. You could also check the voltage the other side
> of the series resistors to confirm it is the same. If so it looks like a
> software bug perhaps in setting up the ADC. I am not familiar with the
> boards or either firmware.

Is there a way we can leverage the proper behavior of Marlin2 to figure out
what it's doing vs. Klipper on the Archim2?

FHeilmann, 2019-08-26:
I see, apologies for the confusion. Frankly I'm now out of ideas what the problem could be here.

nophead, 2019-08-26:
If you have a digital multimeter you could measure the voltage on the thermistor input and confirm it is correct for a 320 Ohm resistor in a potential divider with 4K7. You could also check the voltage the other side of the series resistors to confirm it is the same. If so it looks like a software bug perhaps in setting up the ADC.  I am not familiar with the boards or either firmware.

nophead, 2019-08-27:
You would need to compare the code.

bryanboettcher, 2019-08-27:
Any suggestions for starting points?  I could maybe fumble my way through
the Python host part of Klipper, but I doubt the firmware core will do me
any good ... and that's probably where any bug is.

KevinOConnor, 2019-09-06:
I'm going to close this issue as a fix was implemented for the original problem report.

-Kevin

closed 2019-09-06 by KevinOConnor