Klipper3d/klipper #1911: Extruder skipping steps when PA active

tgmorris99, 2019-08-25:
I recently installed a Zesty Nimble and when trying to print 0.3 layer height at fast speeds my extruder stepper is skipping steps from time to time. I'm guessing that the PA is trying to pressurize things when this happens. If I turn off PA or significantly reduce the speed I do not have the same issue. If I lower the max_accel from 1080 to 720 the problem also goes away. Additionally, I do not have the problem at all when running 0.2 layers up to 150mm/s.

It seems that the filament acceleration is not being limited by any of the acceleration values when PA is active and the higher flow volume due to the 0.3 layer uncovers the issue.

The attached log is at max_accel of 1080, 0.3 layer height, 120mm/s print speed and has several instances of the extruder stepper skipping steps.

I also tried one of the recommended steppers as they have a lower rotor inertia. While it helped it did not solve the issue.

[klippy.log](https://github.com/KevinOConnor/klipper/files/3537754/klippy.log)

Hywelmartin, 2019-08-25:
#1758

tgmorris99, 2019-08-25:
It does seem to be the same issue so I'll close this and add to the other.

closed 2019-08-25 by tgmorris99