Klipper3d/klipper #1912: Homing no longer uses acceleration when starting a movement

Taudris, 2019-08-25:
I believe homing is no longer respecting `max_z_accel` when starting movement. Can this be fixed? I just updated Klipper for the first time in months, and had to reduce my Z axis `homing_speed` from an already-slow 16mm/s to a pitiful 5mm/s in order to get my Z stepper to stop making a high pitched whine and start moving.

closed 2019-09-01 by klipper-gitissuebot

psych0d0g, 2019-09-01:
I have observed this phenomenon aswell.
`Sep 01 18:54:29 alarm python2[2128]: INFO:root:Starting Klippy...
Sep 01 18:54:31 alarm python2[2128]: INFO:root:Start printer at Sun Sep  1 18:54:31 2019 (1567364071.3 3886.7)
Sep 01 18:54:33 alarm python2[2128]: INFO:root:Extruder max_extrude_ratio=0.266081
Sep 01 18:54:33 alarm python2[2128]: INFO:root:Starting serial connect
Sep 01 18:54:34 alarm python2[2128]: INFO:root:Loaded MCU 'mcu' 80 commands (v0.7.0-775-g078d278-20190821_225806-Cronix-PC / gcc: (15:7-2018-q2-6) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907] binutils: (2.31.1-12+11) 2.31.1)
Sep 01 18:54:34 alarm python2[2128]: MCU 'mcu' config: BUS_PINS_twi1=PB5,PB4 BUS_PINS_twi0=PA4,PA3 BUS_PINS_spi=PA12,PA13,PA14 STEP_DELAY=2 BUS_PINS_usart0=PB0,PB1,PB13 ADC_MAX=4095 BUS_PINS_usart1=PA21,PA22,PA23 PWM_MAX=255 CLOCK_FREQ=120000000 RESERVE_PINS_USB=PB10,PB11 MCU=sam4e8e STATS_SUMSQ_BASE=256
Sep 01 18:54:34 alarm python2[2128]: INFO:root:Sending MCU 'mcu' printer configuration...
Sep 01 18:54:34 alarm python2[2128]: WARNING:root:got {'#receive_time': 3889.347773881, u'crc': 2682785996, u'is_shutdown': 0, '#name': u'config', u'is_config': 1, '#sent_time': 3889.346721538, u'move_count': 1024}
Sep 01 18:54:34 alarm python2[2128]: INFO:root:Configured MCU 'mcu' (1024 moves)
Sep 01 18:54:34 alarm python2[2128]: INFO:root:Starting heater checks for heater_bed
Sep 01 18:54:34 alarm python2[2128]: INFO:root:Starting heater checks for extruder0
Sep 01 18:54:34 alarm python2[2128]: INFO:root:Stats 3889.4: gcodein=0 mcu: mcu_awake=0.000 mcu_task_avg=0.000000 mcu_task_stddev=0.000000 bytes_write=1567 bytes_read=3728 bytes_retransmit=9 bytes_invalid=0 send_seq=167 receive_seq=167 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=120190381 heater_bed: target=0 temp=0.0 pwm=0.000 print_time=9.681 buffer_time=0.249 print_stall=0 extruder0: target=0 temp=0.0 pwm=0.000
Sep 01 18:54:35 alarm python2[2128]: INFO:root:Stats 3890.4: gcodein=0 mcu: mcu_awake=0.004 mcu_task_avg=0.000009 mcu_task_stddev=0.000016 bytes_write=1573 bytes_read=3789 bytes_retransmit=9 bytes_invalid=0 send_seq=168 receive_seq=168 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=120004025 heater_bed: target=0 temp=27.9 pwm=0.000 print_time=9.681 buffer_time=0.000 print_stall=0 extruder0: target=0 temp=27.7 pwm=0.000
Sep 01 18:54:35 alarm python2[2128]: INFO:root:Unknown command:"M117 192.168.1.151 3344"
Sep 01 18:54:36 alarm python2[2128]: INFO:root:Stats 3891.4: gcodein=123 mcu: mcu_awake=0.004 mcu_task_avg=0.000009 mcu_task_stddev=0.000016 bytes_write=1579 bytes_read=3925 bytes_retransmit=9 bytes_invalid=0 send_seq=169 receive_seq=169 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=119995488 heater_bed: target=0 temp=27.8 pwm=0.000 print_time=9.681 buffer_time=0.000 print_stall=0 extruder0: target=0 temp=27.6 pwm=0.000
Sep 01 18:54:37 alarm python2[2128]: INFO:root:Stats 3892.4: gcodein=134 mcu: mcu_awake=0.004 mcu_task_avg=0.000009 mcu_task_stddev=0.000016 bytes_write=1585 bytes_read=4031 bytes_retransmit=9 bytes_invalid=0 send_seq=170 receive_seq=170 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=119994047 heater_bed: target=0 temp=27.8 pwm=0.000 print_time=9.681 buffer_time=0.000 print_stall=0 extruder0: target=0 temp=27.7 pwm=0.000
Sep 01 18:54:38 alarm python2[2128]: INFO:root:Stats 3893.4: gcodein=145 mcu: mcu_awake=0.004 mcu_task_avg=0.000009 mcu_task_stddev=0.000016 bytes_write=1591 bytes_read=4137 bytes_retransmit=9 bytes_invalid=0 send_seq=171 receive_seq=171 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=119995049 heater_bed: target=0 temp=27.9 pwm=0.000 print_time=9.681 buffer_time=0.000 print_stall=0 extruder0: target=0 temp=27.6 pwm=0.000
Sep 01 18:54:39 alarm python2[2128]: INFO:root:Stats 3894.4: gcodein=157 mcu: mcu_awake=0.004 mcu_task_avg=0.000009 mcu_task_stddev=0.000016 bytes_write=1597 bytes_read=4273 bytes_retransmit=9 bytes_invalid=0 send_seq=172 receive_seq=172 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=119994429 heater_bed: target=0 temp=27.9 pwm=0.000 print_time=9.681 buffer_time=0.000 print_stall=0 extruder0: target=0 temp=27.7 pwm=0.000
Sep 01 18:54:40 alarm python2[2128]: INFO:root:Stats 3895.4: gcodein=169 mcu: mcu_awake=0.001 mcu_task_avg=0.000005 mcu_task_stddev=0.000005 bytes_write=1603 bytes_read=4393 bytes_retransmit=9 bytes_invalid=0 send_seq=173 receive_seq=173 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=119995029 heater_bed: target=0 temp=27.8 pwm=0.000 print_time=9.681 buffer_time=0.000 print_stall=0 extruder0: target=0 temp=27.7 pwm=0.000
Sep 01 18:54:41 alarm python2[2128]: INFO:root:Stats 3896.4: gcodein=181 mcu: mcu_awake=0.001 mcu_task_avg=0.000005 mcu_task_stddev=0.000005 bytes_write=1609 bytes_read=4499 bytes_retransmit=9 bytes_invalid=0 send_seq=174 receive_seq=174 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=119995160 heater_bed: target=0 temp=27.9 pwm=0.000 print_time=9.681 buffer_time=0.000 print_stall=0 extruder0: target=0 temp=27.7 pwm=0.000
Sep 01 18:54:42 alarm python2[2128]: INFO:root:Stats 3897.4: gcodein=208 mcu: mcu_awake=0.001 mcu_task_avg=0.000005 mcu_task_stddev=0.000005 bytes_write=2115 bytes_read=4730 bytes_retransmit=9 bytes_invalid=0 send_seq=189 recei [... cut ...] 0: target=0 temp=27.6 pwm=0.000
Sep 01 18:55:20 alarm python2[2128]: INFO:root:Stats 3935.4: gcodein=656 mcu: mcu_awake=0.001 mcu_task_avg=0.000005 mcu_task_stddev=0.000005 bytes_write=13959 bytes_read=11261 bytes_retransmit=9 bytes_invalid=0 send_seq=584 receive_seq=584 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=119993983 heater_bed: target=0 temp=27.9 pwm=0.000 print_time=61.197 buffer_time=5.717 print_stall=0 extruder0: target=0 temp=27.7 pwm=0.000
`

KevinOConnor, 2019-09-01:
Should be fixed now (commit 05edd556).

-Kevin