Klipper3d/klipper #1913: Anycubic Kossel Linear Plus Parse max temp error

ChrisS1986, 2019-08-25:
Hello, i have a Problem with a Anycubic Kossel Linear Plus. 

Unable to parse option 'max_temp' in section 'extruder'

Since a Update from Octoprint i dont Works anymore. 
I Updatet Klipper and the Microcontroller.
I Write the Config new, but nothing helps

[klippy.log](https://github.com/KevinOConnor/klipper/files/3538496/klippy.log)

BlackStump, 2019-08-25:
you are missing [homing_override] <==========
max_temp = 250

<===========
	G90   ; Uncomment these 2 lines to blindly lift the Z 2mm at start
	G1 Z8 F600
	G28 X0 Y0
	;G1 X234 Y215 F3600
	G1 X91.5 Y161 F3600
	G4 P500 ; wait 1 s
	G28 Z0
	G91
	G1 Z5 F600
	G90

look here for homing override details
https://github.com/KevinOConnor/klipper/blob/master/config/sample-probe-as-z-endstop.cfg#L24

KevinOConnor, 2019-09-06:
I'm going to close this issue as it appears the original question was answered (the config sections were inadvertently altered to an invalid state).

-Kevin

closed 2019-09-06 by KevinOConnor