Klipper3d/klipper #1915: Extruder step-distance per filament.

QuantusHoka, 2019-08-26:
I calibrate my extruder for each new roll of filament, and there can be as much as 15% difference between filaments (PC+ has much higher e-steps / lower step-distance than PLA, for example). With Marlin, I can add custom gcode to the slicer filament settings to set the steps for a given filament, but I can't find a similar function in Klipper. Am I missing something? Does the M221 flow percentage setting functionally accomplish the same thing?

Thanks.

AndyWo83, 2019-08-26:
Hello @QuantusHoka ,
How do you calibrate your extruder?
It is important that you calibrate your extruder without connected bowden, so the conveyed length is always the same, as there is no back pressure from the Nozzle.
The sensitivity you must set in the slicer (Cura, Prusa Slicer, etc.) by flow!

FHeilmann, 2019-08-26:
As andy already mentioned, you can also set this in the slicer bei adjusting the extrusion multiplier. The slicer is the preferred location for making these adjustments, since creating new profiles for new filaments is easily done.

FHeilmann, 2019-08-26:
> Out of curiosity, does anyone know how the flow percentage setting is used? Is the nominal calculated steps for a given move simply multiplied by the flow percentage to produce the steps sent to the printer?

M221 is used to adjust flow while the printer is printing. What the slicer does is it calculates the required amount of filament based on the geometry of the line that has to be put down, and multiplies that value with the extrusion mutliplier. Is if your extruder should move, say, 1mm during a given move, setting the extrusion multiplier to 0.90 will cause the slicer to generate a move like, e.g. `G1 X... Y... E0.90 F...` instead. The firmware can remain at 100% flow rate this way and you only need to adjust your slicer to the multiplier you determined using your calibration prints.

dewi-ny-je, 2020-01-07:
The ticket has been closed but the question has not been answered.
Is it possible to change the extruder calibration from the gcode? The workaround of using the multiplier when, as explained and as well known, the actual filament length changes depending on the softness is that, a workaround.
In theory the slicer should set the correct value per filament.