Klipper3d/klipper #1917: Heated bed switched off automatically

mapa3m1, 2019-08-26:
Hello! Whe i set some tempurature for heated bed, for example 90 celsius, printer reach this tempertature, and it waiting a few min and switch off heating.
But it should keep 90 deg. for drying filament) 
Please advice how to diable auto switchin off

BlackStump, 2019-08-26:
https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L483

mapa3m1, 2019-08-26:
> https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L483

it's look like something different. 
Or i should increase time?

mapa3m1, 2019-08-26:
> https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L483

ohh. i remove TURN_OFF_HEATERS from  DEFAULT_IDLE_GCODE, let's see

Hywelmartin, 2019-08-26:
It can be done with "gcode"
SET_IDLE_TIMEOUT [TIMEOUT=<timeout>]: Allows the user to set the idle timeout (in seconds).

Ph0rkoz, 2019-08-28:
womp womp, that explains why my bed was turning off when preheating it. Takes a good 5 or so minutes to reach 115C so I tend to do other things while it heats, eg slice the model.

closed 2019-09-02 by klipper-gitissuebot