Klipper3d/klipper #1918: open drain for smoothieboard

Akex2, 2019-08-26:
hi
ATM is'nt possible to use open drain mode on smoothie board, 
i can't use my external driver :( it is possible to add this feature please ?

Akex2, 2019-08-26:
add log
[klippy.log](https://github.com/KevinOConnor/klipper/files/3541204/klippy.log)

arthurwolf, 2019-08-26:
Hey.

In case this can help, the code is pretty simple: https://github.com/Smoothieware/Smoothieware/blob/edge/src/libs/Pin.cpp#L98
There are other options you can add btw, see the same file.

Cheers.

Akex2, 2019-08-29:
hi Kevin
open drain it used ex: for control a driver motor (5v input, step,dir,enable) with a 3.3v microcontroler
see this link for more information: http://smoothieware.org/general-appendixes#external-stepper-drivers Watch the second picture
thx

KevinOConnor, 2019-09-06:
Okay, I understand.  This change would require both micro-controller code changes as well as host code changes.  So, a developer will need to step forward to implement this.

-Kevin

KevinOConnor, 2019-10-08:
I'm going to mark this as inactive as it does not appear it is currently in development.

-Kevin

closed 2019-10-08 by KevinOConnor