Klipper3d/klipper #1919: Timeout with MCU 'mcu'

ushills, 2019-08-26:
I've had a number of failures when printing where the connection to the mcu appears to be lost.

I have the recommended power supply, a ground isolated USB cable (tried two new ones), however, randomly the print will fail due to lost communication with the mcu.

The corresponding event is syslog appears to be:

`
Aug 26 18:13:34 octopi kernel: [69557.397689] ch341-uart ttyUSB1: usb_serial_generic_read_bulk_callback - urb stopped: -32`

Any ideas would be appreciated as I'm loving Klipper otherwise.

[klippy.log.gz](https://github.com/KevinOConnor/klipper/files/3542261/klippy.log.gz)

ushills, 2019-08-26:
Would dropping to  USB 1.1 as described here rather than 2.0 cause any problems with Klipper and baud rate. 

https://github.com/raspberrypi/linux/issues/1187#issuecomment-154383134

ushills, 2019-08-26:
Having tried reducing usb speed to 1.1 made no difference and next print also failed.

ushills, 2019-08-27:
Mainboard is a Geeetech GT2560 Rev A+, the USB cable is 0.9m, thick and shielded and has clamp-on toroids both ends.  Raspberry PI 3B, with OctoPi and Klipper only installed, I saw lots of mjpg and usb-camera errors in the log, but I don't have a USB camera attached and have disabled the feature in OctoPrint.

ushills, 2019-08-27:
Just before failure, the bytes_retransmit increased massively from the average of 9 to 371, 914 to 1095 and a number of stalled_bytes.  The issue seems to definitely be a data transmission issue from the RPi to the CH341 uart convertor.

`Stats 69557.6: gcodein=46304 mcu: mcu_awake=0.073 mcu_task_avg=0.000229 mcu_task_stddev=0.000222 bytes_write=8317756 bytes_read=1920022 bytes_retransmit=371 bytes_invalid=0 send_seq=167583 receive_seq=167580 retransmit_seq=167583 srtt=0.004 rttvar=0.000 rto=0.100 ready_bytes=21 stalled_bytes=153 freq=16000283 heater_bed: target=45 temp=45.0 pwm=0.275 sd_pos=2820023 print_time=14584.229 buffer_time=2.065 print_stall=0 extruder: target=190 temp=190.0 pwm=0.307
Stats 69558.6: gcodein=46308 mcu: mcu_awake=0.073 mcu_task_avg=0.000229 mcu_task_stddev=0.000222 bytes_write=8317756 bytes_read=1920022 bytes_retransmit=914 bytes_invalid=0 send_seq=167583 receive_seq=167580 retransmit_seq=167583 srtt=0.004 rttvar=0.000 rto=0.800 ready_bytes=21 stalled_bytes=1574 freq=16000283 heater_bed: target=45 temp=45.0 pwm=0.275 sd_pos=2821365 print_time=14585.374 buffer_time=2.208 print_stall=0 extruder: target=190 temp=190.0 pwm=0.307
Stats 69559.6: gcodein=46312 mcu: mcu_awake=0.073 mcu_task_avg=0.000229 mcu_task_stddev=0.000222 bytes_write=8317756 bytes_read=1920022 bytes_retransmit=1095 bytes_invalid=0 send_seq=167583 receive_seq=167580 retransmit_seq=167583 srtt=0.004 rttvar=0.000 rto=1.600 ready_bytes=21 stalled_bytes=2833 freq=16000283 heater_bed: target=45 temp=45.0 pwm=0.275 sd_pos=2822494 print_time=14586.229 buffer_time=2.063 print_stall=0 extruder: target=190 temp=190.0 pwm=0.307
Stats 69560.6: gcodein=46321 mcu: mcu_awake=0.073 mcu_task_avg=0.000229 mcu_task_stddev=0.000222 bytes_write=8317756 bytes_read=1920022 bytes_retransmit=1095 bytes_invalid=0 send_seq=167583 receive_seq=167580 retransmit_seq=167583 srtt=0.004 rttvar=0.000 rto=1.600 ready_bytes=21 stalled_bytes=4194 freq=16000283 heater_bed: target=45 temp=45.0 pwm=0.275 sd_pos=2823807 print_time=14587.286 buffer_time=2.120 print_stall=0 extruder: target=190 temp=190.0 pwm=0.307
Timeout with MCU 'mcu' (eventtime=69561.602799)`

ushills, 2019-08-27:
I have reduced the baud rate to 115200 and adjusted the [printer] parameters and I am currently printing my second benchy and the bytes_retransmit is equal to zero.  Which is the lowest it has been since using Klipper.

I will run a few more complicated prints and report back if that solves the issue.

ushills, 2019-08-28:
> another route that is guaranteed to work is consider the SKR 1.3 mainboard ugprade.

What is the difference with the SKR 1.3 other than 32 bit, I assume I still need to connect via USB from the RPi.  Is there a board with the RPi or equivalant as part of the board, that can run the Klipper software and firmware.

BlackStump, 2019-08-28:
I have a Geeetech 2560 A+ connected to a Orangepi Zero Plus and do not have any comms issues.
I have been using it in a Ender3 for well over a year.
nothing fancy as far as usb cable. I dont use Octoprint tho.

ushills, 2019-08-29:
Reducing the baud to 115200 appears to have resolved the issues (115200 is the standard for the Geeetech GT2560A+).

I am no longer seeing any dropped bytes in the log file and the last test prints have all worked even when increasing print speed (x2).

closed 2019-08-29 by ushills