Klipper3d/klipper #1920: Stepper motor current - MKS Sbase 1.3

Grorbak, 2019-08-28:
Hi,

I'm looking for install klipper on my MKS Sbase 1.3

But i don't find how to adjust current on the config file for the inboard stepper driver.

The board use DRV8825 without pin to adjust any voltage.

Did you have a command line to add for pull the driver the good current ?

Thanks a lot for your answer.

Hywelmartin, 2019-08-28:
isn't that board a smoothie clone??
So have a look here.
https://github.com/KevinOConnor/klipper/blob/master/config/generic-smoothieboard.cfg

Grorbak, 2019-08-28:
Is it a smoothie clone, but in the generic file, they don't talk about current on it.
That's why i'm asking here ^^

Hywelmartin, 2019-08-28:
https://github.com/KevinOConnor/klipper/blob/c95209bf7aeab1fa8ea867d85645a2171a5db590/config/generic-smoothieboard.cfg#L87

I would assume

Grorbak, 2019-08-28:
Ooooooh that's it.

Ok, thank you so much !

closed 2019-08-28 by Grorbak