Klipper3d/klipper #1921: Probed values deviation using BLTouch

jlirochon, 2019-08-28:
I have a weird issue with my genuine BLTouch v3.1 on an Ender 3 (I am aware of the known capacitor problems on Melzi boards, but BLTouch is connected to another pin) 

When I issue a `PROBE_ACCURACY` command, probed Z values keep increasing. Delta between 2 probes varies, but the probed value is always greater than the previous.

I tested with 40 probes on the same X,Y point, the first value is `z=0.930500` and the last is `z=1.360500`.

After homing Z again, it restarts at `z=0.930500` then keeps increasing again. Since the homing constantly occurs around `z=0.930500` I suppose the BLTouch itself performs good, and that the deviation comes from a configuration problem or a bug ?

```
Recv: // probe at 158.000,134.000 is z=0.930500
Recv: // probe at 158.000,134.000 is z=0.938000
(...)
Recv: // probe at 158.000,134.000 is z=1.353000
Recv: // probe at 158.000,134.000 is z=1.360500
```

Let me know if I can do something to help !
Thank you

[klippy.log](https://github.com/KevinOConnor/klipper/files/3552791/klippy.log)
[octoprint.log](https://github.com/KevinOConnor/klipper/files/3552828/octoprint.log)
[printer.txt](https://github.com/KevinOConnor/klipper/files/3552909/printer.txt)

KevinOConnor, 2019-08-29:
Your Klipper log file is incomplete (or you've done a custom install and omitted the "-l" command line option - which should be avoided as logging could then lead to timing errors in the host ).  We'll need to see a complete Klipper log file in order to assist.

-Kevin

ushills, 2019-08-29:
Reducing the baud rate in the firmware and printer.cfg to 115200 appears to have solved the problem.  115200 is the default with the GT2560A+ board so I suppose this makes sense.

All working now and I'm seeing no errors (dropped bytes) in the klippy.log file.

jlirochon, 2019-08-29:
Yes it's a custom install with [supervisor](http://supervisord.org/) redirecting `stderr` and `stdout` to a log file.

Here is the output with the `-l` option. I issued `M112` right after the 40 probes : [klippy2.log](https://github.com/KevinOConnor/klipper/files/3556507/klippy2.log)

jlirochon, 2019-09-01:
The deviation was due to a mechanical problem 😳

During my tests I once slightly crashed the nozzle, and it seems Z axis rollers came loose.

I disassembled the printed and adjusted it again, and now probing is accurate.

closed 2019-09-01 by jlirochon