Klipper3d/klipper #1922: Move exceeds maximum extrusion error

oullah, 2019-08-29:
I just installed Klipper on my Ender 3 and was trying to do my first print. As soon as heating is complete and it tries to lay down the first filament I am getting the maximum extrusion error: 

```
Send: N19 G1 X110.347 Y118.368 E1.00304*89
Recv: // Move exceeds maximum extrusion (29.785mm^2 vs 0.640mm^2)
Recv: // See the 'max_extrude_cross_section' config option for details
Recv: !! Move exceeds maximum extrusion (29.785mm^2 vs 0.640mm^2)
```
I'm using Cura to slice the file. Perhaps I have a bad setting somewhere that worked fine with the stock firmware? 

Log file: [klippy.zip](https://github.com/KevinOConnor/klipper/files/3553929/klippy.zip)

Hywelmartin, 2019-08-29:
Probably from an excessive purge line. That can be altered in the start gcode..

oullah, 2019-08-29:
This is the only stat gcode: 
```
; Ender 3 Custom Start G-code
G28 ; Home all axes
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
; End of custom start GCode
```
I actually removed the purge line and had the issue, when I put the purge line back it printed the purge line and then threw the same error when trying to start the print. 

Could it be retraction settings? I don't see `max_extrude_cross_section` listed on the [slicers](https://github.com/KevinOConnor/klipper/blob/master/docs/Slicers.md) page but it seems like it could be a culprit...

jlirochon, 2019-08-29:
I had a similar issue with my Ender 3 using Slic3r. I adapted the default start gcode.

Find the last gcode command printed before the error, look at the `E` parameter, for example in `G1 X0.4 Y20 Z0.3 F1500.0 E30` it is `E30`. The error message contains the actual extrusion and the max extrusion. You can compute a new `E` value with simple maths. In my case it fixed the problem, it was a purge line. The stepper was losing steps and producing an horrible noise during the purge on Marlin, it's clear that the value was too high. I'm glad Klipper helped to fix it.

Otherwise, if you are sure you should extrude that much, you can set `max_extrude_cross_section` to a higher value in your config.

By the way there is a huge difference between the 2 values. Either you are extruding much without moving, or you have a micro nozzle (not to offend you)

jlirochon, 2019-08-29:
The default is `4.0 * nozzle_diameter^2` then it is `0.640mm^2`for a 0.4 nozzle (your nozzle is within the norm) 

`29.785mm^2` seems a huge value. The stepper of my Ender3 was losing steps at values like `0.722mm^2`. Your gcode probably extrudes to much without moving the head. Don't you hear a pneumatic drill sound when you run this gcode on Marlin ?

oullah, 2019-08-29:
Here is the GCODE (after the customer start code and up to the error): 
```
G92 E0
G1 F1800 E-4
;LAYER_COUNT:319
;LAYER:0
M107
;MESH:3DBenchy.stl
G0 F9000 X110.347 Y118.449 Z0.2
;TYPE:WALL-INNER
G1 F600 E0
;FLOW_RATE_COMPENSATED_OFFSET = 1.000000
G1 F1800 X110.347 Y118.368 E1.00304
```
It looks like I had forgotten to turn off an experimental feature of `FLOW_RATE_COMPENSATED_OFFSET`. The extruder values seem to make more sense now that I turned that down to 0. Looks like its printing now, so I guess I just need to tune my steps and pressure advance. Thanks for your help!

closed 2019-08-29 by oullah