Klipper3d/klipper #1923: Option 'set_servo servo' is not valid in section 'probe'

x1a999, 2019-08-29:
Ender 3 with Creality 1.1.3 board.

I have the Z Endstop servo connected to the Pin 27 breakout board (of the LCD)

Problem is that when I restart the Klipper host (in OctoPrint) it's showing me:

`Option 'set_servo servo' is not valid in section 'probe'`

This is my printer.cfg file:

`
# This file contains common pin mappings for the 2018 Creality
# Ender 3. To use this config, the firmware should be compiled for the
# AVR atmega1284p.

# Note, a number of Melzi boards are shipped with a bootloader that
# requires the following command to flash the board:
#  avrdude -p atmega1284p -c arduino -b 57600 -P /dev/ttyUSB0 -U out/klipper.elf.hex
# If the above command does not work and "make flash" does not work
# then one may need to flash a bootloader to the board - see the
# Klipper docs/Bootloaders.md file for more information.

# See the example.cfg file for a description of available parameters.

[stepper_x]
step_pin: PD7
dir_pin: !PC5
enable_pin: !PD6
step_distance: .0125
endstop_pin: ^PC2
position_endstop: 0
position_max: 235
homing_speed: 50

[stepper_y]
step_pin: PC6
dir_pin: !PC7
enable_pin: !PD6
step_distance: .0125
endstop_pin: ^PC3
position_endstop: 0
position_max: 235
homing_speed: 50

[stepper_z]
step_pin: PB3
dir_pin: PB2
enable_pin: !PA5
step_distance: .0025
endstop_pin:probe:z_virtual_endstop
position_endstop: 15
position_max: 250

[homing_override]
set_position_z: 0
gcode:
 G90 
 G1 Z30 F600 ; Blindly lift the Z 30mm at start 
 G28 X0 Y0 
 G1 X88 Y145 F3600 
 G28 Z0
 G1 Z10 F1000

[servo EZABL]
pin: PA4
maximum_servo_angle: 180
minimum_pulse_width: 0.001
maximum_pulse_width: 0.002
initial_angle: 35
enable: true

[probe]
pin: ^!PC4
z_offset: 15.0
activate_gcode:
set_servo servo=EZABL ENABLE=1
    set_servo servo=EZABL ANGLE=35
    G4 P1000
deactivate_gcode:
set_servo servo=EZABL ANGLE=5
    set_servo servo=EZABL ENABLE=0
    G4 P500

[bed_tilt]
horizontal_move_z: 15
#x_adjust:
#y_adjust:
points:
    5,65
    85,65
    165,65
    165,145
    85,145
    5,145
    5,215
    85,215
    165,215
	
[extruder]
max_extrude_only_distance: 100.0
step_pin: PB1
dir_pin: !PB0
enable_pin: !PD6
#step_distance: 0.010526
step_distance: 0.003509
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PD5
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PA7
control: pid
# tuned for stock hardware with 200 degree Celsius target
pid_Kp: 21.527
pid_Ki: 1.063
pid_Kd: 108.982
min_temp: 0
max_temp: 250

[heater_bed]
heater_pin: PD4
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PA6
control: pid
# tuned for stock hardware with 50 degree Celsius target
pid_Kp: 54.027
pid_Ki: 0.770
pid_Kd: 948.182
min_temp: 0
max_temp: 90

[fan]
pin: PB4

[mcu]
serial: /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A107PKOB-if00-port0

[printer]
kinematics: cartesian
max_velocity: 300
max_accel: 3000
max_z_velocity: 5
max_z_accel: 100

[display]
lcd_type: st7920
cs_pin: PA3
sclk_pin: PA1
sid_pin: PC1
encoder_pins: ^PD2, ^PD3
click_pin: ^!PC0
`

x1a999, 2019-08-29:
Thanks automated script... I solved it :D

closed 2019-08-29 by x1a999

JungDev, 2021-01-23:
How you solve? I have same error