Klipper3d/klipper #1924: CR-10s Z-offset

beauwinden, 2019-08-30:
First, I just want to say thank you to KevinOConnor and anyone else who has worked on Klipper. It's been great so far.

This previous issue was closed, but I wanted to add my thoughts about it.

https://github.com/KevinOConnor/klipper/issues/723

I have a Creality CR-10s, came from stock firmware and set up Octoprint + Klipper. I don't have automatic bed leveling. My bed was recently leveled right before the firmware update, using the paper test in multiple areas of the bed and getting just a slight bit of tension. I get a good "squish" on the first layer without scratching the bed.

I used the CR-10s Klipper config profile, assuming it was a good starting point. Edited only the mcu area and figured I would try a test print after I verified that Klipper was properly controlling my stepper motors, temperature sensors, etc. I did the full tests from the Config Checks page.

I loaded up the stl for a configuration cube that I printed last night with the old firmware without a hitch. The print started, and it took me longer than I'd like to admit to realize that there was a problem.

Due to the 0.5 z-offset setting, it dug a very deep configuration cube sized square into my bed, plus the tool path to get there and the skirt. I'm using an expensive powder coated PEI spring steel magnetic build plate and the damage is unfixable - all I can do is flip it over.

I'm at fault for not noticing it sooner, but it shouldn't have come to that. Even after setting my z-offset to 0.0, I still had to lower the bed a bit more and re-level. This really needs to at least be changed to 0.0 to minimize the chances of damage, but maybe even to -0.2. I'd rather get bad adhesion at first and have to raise my bed or change this setting than dig a hole in my bed.

I know Kevin said he doesn't have this printer and didn't create the config file. However, I do have this printer and I was affected by the decision made on the other issue to leave the config file alone. I'm probably not the first person to damage their bed using this file, and if this isn't changed, I won't be the last.

Hywelmartin, 2019-08-30:
The whole idea to have the endstop at Z=0.5 is to avoid crashing into the bed during the homing cycle..

A manual bed level should be sufficient to get everything in order,   before the first test print ( where you tell the printer to go to Z0.0 not the home Z height). I'm not certain if this is totally clear in the documentation...

KevinOConnor, 2019-08-31:
Okay, I changed the Z position_endstop=0 in the example Creality configs (commit f7f6680b).

-Kevin

closed 2019-09-14 by KevinOConnor