Klipper3d/klipper #1925: Cant home Z axis and bltouch is not working

felix900006, 2019-08-30:
Hello,
i have a Ender 5 with Klipper.
I got the Problem that i cant home the Z axis if i try it the XY Axis Home and the Z axis does not move at all also when i set Z to 0 and then home it moves but it doesnt stop and crashes into the extruder but the BLtouch is triggert.
[klippy.log](https://github.com/KevinOConnor/klipper/files/3561119/klippy.log)

MalsakerLures, 2019-08-30:
I think you are missing a Homing Overide section:
[homing_override]
gcode:
  
    G90 ; Use absolute position mode
    G1 Z15 ; Move up 10mm
    G28 X Y
    G1 X144.25 Y114.63 F6000 
    G28 Z
set_position_z: 0.0

felix900006, 2019-08-31:
what in my config is:

[homing_override]
gcode:
    G90 ; Use absolute position mode
    G1 Z10 ; Move up 10mm
    G28 X Y
    G1 X166 Y120 F6000 ; Change the X and Y coordinates to the center of your print bed
    G28 Z
set_position_z: 0.0

felix900006, 2019-09-01:
i just looked in to the Log file and found this Error

`BLTouch failed to verify sensor state
Script running error
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/extras/display/menu.py", line 1539, in dispatch_gcode
    self.gcode.run_script(script)
  File "/home/pi/klipper/klippy/gcode.py", line 292, in run_script
    self._process_commands(script.split('\n'), need_ack=False)
  File "/home/pi/klipper/klippy/gcode.py", line 224, in _process_commands
    handler(params)
  File "/home/pi/klipper/klippy/gcode.py", line 74, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
  File "/home/pi/klipper/klippy/extras/homing_override.py", line 60, in cmd_G28
    self.template.run_gcode_from_command(kwparams)
  File "/home/pi/klipper/klippy/extras/gcode_macro.py", line 68, in run_gcode_from_command
    self.gcode.run_script_from_command(self.render(context))
  File "/home/pi/klipper/klippy/gcode.py", line 287, in run_script_from_command
    self._process_commands(script.split('\n'), need_ack=False)
  File "/home/pi/klipper/klippy/gcode.py", line 224, in _process_commands
    handler(params)
  File "/home/pi/klipper/klippy/gcode.py", line 74, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
  File "/home/pi/klipper/klippy/extras/homing_override.py", line 22, in cmd_G28
    self.gcode.cmd_G28(params)
  File "/home/pi/klipper/klippy/gcode.py", line 532, in cmd_G28
    homing_state.home_axes(axes)
  File "/home/pi/klipper/klippy/homing.py", line 126, in home_axes
    self.toolhead.get_kinematics().home(self)
  File "/home/pi/klipper/klippy/kinematics/cartesian.py", line 82, in home
    self._home_axis(homing_state, axis, self.rails[axis])
  File "/home/pi/klipper/klippy/kinematics/cartesian.py", line 69, in _home_axis
    homing_state.home_rails([rail], forcepos, homepos)
  File "/home/pi/klipper/klippy/homing.py", line 99, in home_rails
    self.homing_move(movepos, endstops, hi.speed)
  File "/home/pi/klipper/klippy/homing.py", line 42, in homing_move
    mcu_endstop.home_prepare()
  File "/home/pi/klipper/klippy/extras/bltouch.py", line 144, in home_prepare
    self.test_sensor()
  File "/home/pi/klipper/klippy/extras/bltouch.py", line 139, in test_sensor
    "verify sensor state")
  File "/home/pi/klipper/klippy/extras/bltouch.py", line 101, in verify_state
    raise homing.EndstopError("BLTouch failed to %s" % (msg,))
EndstopError: BLTouch failed to verify sensor state`

What does it mean?

BlackStump, 2019-09-01:
Have you read the BL-touch troubleshooting section?
https://www.klipper3d.org/BLTouch.html

felix900006, 2019-09-03:
Ok it is working not mostly. The Option `pin_up_touch_mode_reports_triggered: False` did the trick but now i have the problem that the Probe goes over the heatbed when i try to calibrate.

felix900006, 2019-09-04:
Ok i have got the BLToucht to work fine but the only Problem that is remaining is that i cant get it low enough that it prints on the headbed it is always 3-4 mm above the headbed. changing the Z offset did not work.

aegelsky, 2019-09-14:
> Ok i have got the BLToucht to work fine but the only Problem that is remaining is that i cant get it low enough that it prints on the headbed it is always 3-4 mm above the headbed. changing the Z offset did not work.

@felix900006 , did you try adding position_min: -10 in stepper_z ?
looks like it needs to go down more than you have there (-1)

KevinOConnor, 2019-09-29:
I'm going to close this issue as it looks like the original issue was solved and the conversation has concluded on the additional issues.

-Kevin

closed 2019-09-29 by KevinOConnor