Klipper3d/klipper #1926: Unable to get serial connection to cheetah board v1.1 to work

EnnajimiYassine, 2019-08-30:
I'm unable to get serial connection with my cheetah board v1.1 (stm32F1)
My menuconfig : 
                                                                                                       │  
  │ ┌────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────┐ │  
  │ │                         [*] Enable extra low-level configuration options                                               │ │  
  │ │                             Micro-controller Architecture (STMicroelectronics STM32F1/F4)  --->                        │ │  
  │ │                             Processor model (STM32F103)  --->                                                          │ │  
  │ │                             Bootloader offset (28KiB bootloader)  --->                                                 │ │  
  │ │                             Clock Reference (8Mhz crystal)  --->                                                       │ │  
  │ │                         [ ] Use USB for communication (instead of serial)                                              │ │  
  │ │                             Serial Port (USART1)  --->                                                                 │ │  
  │ │                         (250000) Baud rate for serial port                                                             │ │  
  │ │                         [ ] Specify a custom step pulse duration                                                       │ │  
  │ │                         ()  GPIO pins to set at micro-controller startup                                               │ │  
  │ │                                                                               

I also tried use usb for communication but same error , can please someone help ?
thanks a lot !
Please find the logs : 
Starting Klippy...
Args: ['/home/pi/klipper/klippy/klippy.py', '/home/pi/printer.cfg', '-l', '/tmp/klippy.log']
Git version: 'v0.7.0-803-g1b454c0'
CPU: 4 core ARMv7 Processor rev 4 (v7l)
Python: '2.7.13 (default, Sep 26 2018, 18:42:22) \n[GCC 6.3.0 20170516]'
Start printer at Fri Aug 30 22:17:47 2019 (1567199867.7 70.2)
===== Config file =====
[stepper_x]
step_pin = PB8
dir_pin = PB9
enable_pin = !PA8
step_distance = .0125
endstop_pin = ^!PA1
position_endstop = 0
position_max = 200
homing_speed = 50

[tmc2209 stepper_x]
uart_pin = PA3
tx_pin = PA2
uart_address = 0
microsteps = 16
run_current = 0.800
hold_current = 0.500
stealthchop_threshold = 250

[stepper_y]
step_pin = PB2
dir_pin = PB3
enable_pin = !PB1
step_distance = .0125
endstop_pin = ^!PB4
position_endstop = 0
position_max = 200
homing_speed = 50

[tmc2209 stepper_y]
uart_pin = PA3
tx_pin = PA2
uart_address = 1
microsteps = 16
run_current = 0.800
hold_current = 0.500
stealthchop_threshold = 250

[stepper_z]
step_pin = PC0
dir_pin = PC1
enable_pin = !PC2
step_distance = .0025
endstop_pin = ^!PA15
position_endstop = 0.5
position_max = 200

[tmc2209 stepper_z]
uart_pin = PA3
tx_pin = PA2
uart_address = 2
microsteps = 16
run_current = 0.800
hold_current = 0.500
stealthchop_threshold = 5

[extruder]
step_pin = PC15
dir_pin = PC14
enable_pin = !PC13
step_distance = 0.010526
nozzle_diameter = 0.400
filament_diameter = 1.750
heater_pin = PC6
sensor_type = EPCOS 100K B57560G104F
sensor_pin = PC4
control = pid
pid_kp = 21.527
pid_ki = 1.063
pid_kd = 108.982
min_temp = 0
max_temp = 250

[tmc2209 extruder]
uart_pin = PA3
tx_pin = PA2
uart_address = 3
microsteps = 16
run_current = 0.800
hold_current = 0.500
stealthchop_threshold = 5

[heater_bed]
heater_pin = PC7
sensor_type = EPCOS 100K B57560G104F
sensor_pin = PC5
control = pid
pid_kp = 54.027
pid_ki = 0.770
pid_kd = 948.182
min_temp = 0
max_temp = 130

[fan]
pin = PC8

[mcu]
serial = /dev/ttyUSB0

[printer]
kinematics = cartesian
max_velocity = 300
max_accel = 3000
max_z_velocity = 5
max_z_accel = 100

[board_pins]
aliases = 
	EXP1_1=PC9, EXP1_3=PC10,   EXP1_5=PC11, EXP1_7=PB12, EXP1_9=<GND>,
	EXP1_2=PB12, EXP1_4=<RST>, EXP1_6=PB13,  EXP1_8=PB15, EXP1_10=<5V>

[display]
lcd_type = st7920
cs_pin = EXP1_7
sclk_pin = EXP1_6
sid_pin = EXP1_8
encoder_pins = ^EXP1_5, ^EXP1_3
click_pin = ^!EXP1_2

[output_pin beeper]
pin = EXP1_1
=======================
Extruder max_extrude_ratio=0.266081
Starting serial connect
Timeout on serial connect
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/serialhdl.py", line 92, in connect
    identify_data = self._get_identify_data(connect_time + 5.)
  File "/home/pi/klipper/klippy/serialhdl.py", line 57, in _get_identify_data
    params = self.send_with_response(msg, 'identify_response')
  File "/home/pi/klipper/klippy/serialhdl.py", line 156, in send_with_response
    return src.get_response([cmd], self.default_cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 225, in get_response
    raise error("Timeout on wait for '%s' response" % (self.name,))
error: Timeout on wait for 'identify_response' response
Timeout on serial connect
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/serialhdl.py", line 92, in connect
    identify_data = self._get_identify_data(connect_time + 5.)
  File "/home/pi/klipper/klippy/serialhdl.py", line 57, in _get_identify_data
    params = self.send_with_response(msg, 'identify_response')
  File "/home/pi/klipper/klippy/serialhdl.py", line 156, in send_with_response
    return src.get_response([cmd], self.default_cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 225, in get_response
    raise error("Timeout on wait for '%s' response" % (self.name,))
error: Timeout on wait for 'identify_response' response

KevinOConnor, 2019-08-30:
How did you flash the Klipper micro-controller code to the Cheetah?

-Kevin

KevinOConnor, 2019-08-31:
Ah, looks like you were close, but the micro-controller should be compiled for "no bootloader".  Try the steps and config at: https://github.com/KevinOConnor/klipper/blob/work-cheetah-20190830/config/generic-fysetc-cheetah-v1.1.cfg

-Kevin

EnnajimiYassine, 2019-08-31:
Hi kevin, thanks for your quick answer !
I flashed it with the stm32flash command , I'll try with no bootloader and let you know

gsl12, 2019-08-31:
Looks OK to me, but flashing cmd should be "stm32flash -w out/klipper.bin -v -i rts,-dtr,dtr /dev/ttyUSB0". Have a active serial connection, but didn't test any other functions(like LCD, movin stepper g,endstops and so).

EnnajimiYassine, 2019-08-31:
Yes, everything worked (LCD also) but i had to modify the config because the end stops were inverted and I added a bltouch + filament runout 

# This file contains common pin mappings for the Fysetc Cheetah v1.1
# board. To use this config, the firmware should be compiled for the
# STM32F103 with "No bootloader" and with "Use USB for communication"
# disabled.

# The "make flash" command does not work on the Cheetah. Instead,
# after running "make", run the following command to flash the board:
#  stm32flash -w out/klipper.bin -v -i rts,-dtr /dev/ttyUSB0

# See the example.cfg file for a description of available parameters.

[stepper_x]
step_pin: PB8
dir_pin: !PB9
enable_pin: !PA8
step_distance: .0125
endstop_pin: ^PA1
position_endstop: 0
position_max: 200
homing_speed: 50

[tmc2209 stepper_x]
uart_pin: PA3
tx_pin: PA2
uart_address: 0
microsteps: 16
run_current: 0.800
hold_current: 0.500
stealthchop_threshold: 250

[stepper_y]
step_pin: PB2
dir_pin: !PB3
enable_pin: !PB1
step_distance: .0125
endstop_pin: ^PB4
position_endstop: 0
position_max: 200
homing_speed: 50

[tmc2209 stepper_y]
uart_pin: PA3
tx_pin: PA2
uart_address: 1
microsteps: 16
run_current: 0.800
hold_current: 0.500
stealthchop_threshold: 250

[stepper_z]
step_pin: PC0
dir_pin: PC1
enable_pin: !PC2
step_distance: .0025
endstop_pin: probe:z_virtual_endstop
#position_endstop: 0
position_min : -0.55
position_max: 200

[tmc2209 stepper_z]
uart_pin: PA3
tx_pin: PA2
uart_address: 2
microsteps: 16
run_current: 0.800
hold_current: 0.500
stealthchop_threshold: 5

[bltouch]
sensor_pin: PA15
control_pin: PA0
pin_up_reports_not_triggered: False
#   Set if the BLTouch consistently reports the probe in a "not
#   triggered" state after a successful "pin_up" command. This should
#   be True for a genuine BLTouch; some BLTouch clones may require
#   False. The default is True.
pin_up_touch_mode_reports_triggered: False
#   Set if the BLTouch consistently reports a "triggered" state after
#   the commands "pin_up" followed by "touch_mode". This should be
#   True for a genuine BLTouch v2 and earlier; the BLTouch v3 and some
#   BLTouch clones require False. The default is True.
x_offset:-44
y_offset:-6
z_offset:-0.55

[homing_override]
gcode:
    G90 ; Use absolute position mode
    G1 Z10 ; Move up 10mm
    G28 X Y
    G1 X154 Y106 F6000 ; Change the X and Y coordinates to the center of your print bed
    G28 Z
set_position_z: -0.55

[extruder]
step_pin: PC15
dir_pin: !PC14
enable_pin: !PC13
step_distance: 0.010526
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PC6
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PC4
control: pid
pid_kp: 21.527
pid_ki: 1.063
pid_kd: 108.982
min_temp: 0
max_temp: 250

[tmc2209 extruder]
uart_pin: PA3
tx_pin: PA2
uart_address: 3
microsteps: 16
run_current: 1
hold_current: 0.500
stealthchop_threshold: 5

[filament_switch_sensor my_sensor]
pause_on_runout: True
runout_gcode:
	G91
	G1 Z50
	G1 E-40 F500
	M104 S0
	M140 S0
insert_gcode:
	G91
	G1 Z50
	G1 E-40 F500
	M104 S0
	M140 S0
event_delay: 3.0
switch_pin: ^PB5

[heater_bed]
heater_pin: PC7
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PC5
control: pid
pid_kp: 54.027
pid_ki: 0.770
pid_kd: 948.182
min_temp: 0
max_temp: 130

[fan]
pin: PC8

[mcu]
serial: /dev/ttyUSB0

[printer]
kinematics: cartesian
max_velocity: 300
max_accel: 3000
max_z_velocity: 5
max_z_accel: 100

[board_pins]
aliases:
    # EXP1 header
    EXP1_1=PC9,  EXP1_3=PC11, EXP1_5=PC10, EXP1_7=PB12, EXP1_9=<GND>,
    EXP1_2=PC12, EXP1_4=PB14, EXP1_6=PB13, EXP1_8=PB15, EXP1_10=<5V>
    # Pins EXP1_4, EXP1_8, EXP1_6 are also MISO, MOSI, SCK of bus "spi2"

# See the sample-lcd.cfg file for definitions of common LCD displays.
[display]
lcd_type: st7920
cs_pin: EXP1_7
sclk_pin: EXP1_6
sid_pin: EXP1_8
encoder_pins: ^EXP1_5, ^EXP1_3
click_pin: ^!EXP1_2

[output_pin beeper]
pin: EXP1_1

The only problem i have now is that the BLtouch "drop" almost every time after the first touch when homing the Z axis

KevinOConnor, 2019-08-31:
Great.  If you run `ls -l /dev/serial/by-id/` what does it report?

-Kevin

EnnajimiYassine, 2019-08-31:
@KevinOConnor this is the output I have : usb-1a86_USB2.0-Serial-if00-port0
regarding the issue with the BLtouch, do you have any idea please ?

KevinOConnor, 2019-08-31:
Okay, I committed the example configs (commit 5e2c60bd).

> regarding the issue with the BLtouch, do you have any idea please ?

I don't have any experience with the BLTouch myself.  Be sure to follow the steps in https://www.klipper3d.org/BLTouch.html .  If you still having issues after following that guide, then run M112 after the event and attach the klipper log file.

-Kevin

closed 2019-09-06 by klipper-gitissuebot