Klipper3d/klipper #1928: SKR PRO 1.1 TMC2209 UART communication

tiaanv, 2019-09-01:
The UART comms don't work on the SKR PRO 1.1 with TMC 2209 drivers.
The obvious things I have made sure of:

- Power on steppers before startup
- Correct pin mapping (see notes below)
- Steppers & board work without TMC configuration.

I ran BIGTREE's Marlin fork firmware on this board and steppers before this test with success.  The TMC drivers operated 100%, even with Stallguard in use.

Notes:
On doing some research, The SKR PRO config in Marlin used only the RX pin in pin setup for both RX/TX.  This is abnormal. Not sure if they are using a custom implementation custom TMC libraries.
https://github.com/MarlinFirmware/Marlin/issues/14615

[printer_cfg.zip](https://github.com/KevinOConnor/klipper/files/3563190/printer_cfg.zip)
[klippy.log](https://github.com/KevinOConnor/klipper/files/3563187/klippy.log)

KevinOConnor, 2019-09-01:
On the tmc2209 you have to get the uart address correct as well.  What are the MS1/MS2 pins set to?  For example, if they are both pulled high, then it would use uart_address=3.

Otherwise, try with just one driver configured, and if that doesn't work try with just one different driver configured.  Try removing the tx_pin configuration (without it, Klipper will use the uart_pin for both rx and tx).  If that doesn't work, try using just the uart_pin, but with the nominal tx pin for the driver.

-Kevin

tiaanv, 2019-09-03:
Thx Kevin

I tried making the RX & TX pins the same, which of course causes a validation error.

The issue was resolved by ONLY setting the RX pin as you suggested.  Now the UART works 100%

closed 2019-09-03 by tiaanv

KevinOConnor, 2019-09-06:
Thanks.  It sounds like the SKR Pro schematic is not accurate.  As a quick test, what happens if you define "uart_pin" to be the nominal tx pin (and don't configure "tx_pin")?

-Kevin

tiaanv, 2019-09-06:
That's what I did, as per above, and it worked, but with the RX pin?

KevinOConnor, 2019-09-06:
Yes.  If the board follows the schematic, the original config (with both uart_pin and tx_pin) should have worked.  I'm curious in what way the board deviates from the schematic - if uart_pin works with the nominal tx pin then it indicates both rx and tx are wired to the uart but the tx resistor is not present (or its resistance is too low) - if it doesn't work it indicates the tx pin isn't actually wired to the uart.

-Kevin

tiaanv, 2019-09-07:
I will check that for you. Would a simple multimeter test not be able to tell that? Check if both pins are connected, or the impedance.

tiaanv, 2019-09-11:
Was going to do this tonight. But my SKR Pro is DOA. As in toast.  Not sure what or why.   Arm chip heats up... That's never a good sign... Will get back to this when I have a replacement. Sorry

jcm11215, 2019-11-06:
Could you share the working printer.cfg or just the [tmc2209 stepper_x] section and im sure i can figure it out from there. Ive tried every which way i could think of based on the posts here with no luck.

tiaanv, 2019-11-06:
[tmc2209 stepper_y]
uart_pin: PE3
#tx_pin: PE2
microsteps: 16
interpolate: True
run_current: 0.500
hold_current: 0.300
stealthchop_threshold: 100
diag_pin: PE12   # pin on the MCU where DIAG1 is connected (active low)
driver_SGTHRS: 160  # tuning value for sensorless homing, set to 0 as a start

jcm11215, 2019-11-06:
@  tiaanv Thank you 
That got it working. Sensorless homing doesn't work for me though. I have the same diag_pin number as you and i even tried setting the sensitivity to 250. Is sensorless homing working for you?

tiaanv, 2019-11-06:
Yes.

Remember for sensorless homing you need to add:
endstop_pin: tmc2209_stepper_y:virtual_endstop to your stepper config

then I suggest you start at about 140 and play around... remember the sensitivity values are NOT the same as the 2208/2130's.... so play around...

I suggest you start at about

jcm11215, 2019-11-06:
After adding the virtual endstop, all of my axis move in the wrong direction a little and stop. Gives the error Endstop still triggered after retract.

tiaanv, 2019-11-06:
Ok... You should set the retract distance from the R the steppers on x and y to 0.
Did you set the jumpers on the board for the diqg pins to endstops?

jcm11215, 2019-11-06:
Where are the jumpers for that, i dont see them.

tiaanv, 2019-11-06:
Sorry. I am confusing the skr 1.3.  There are a bunch of hardware factors to consider for the sensor less homing. For example... Are you diqg pins soldered in the downward position, and mating with the connector on the board. Either way I think it is beyond the scope of this thread. Happy to help.  You can contact me via email.

jcm11215, 2019-11-06:
Ok that would be great thanks.