Klipper3d/klipper #1929: Bltouch doesn't work with delta printer

Arhigos, 2019-09-01:
So, my printer is Kossel (made by folgertech). I'm using RAMPS 1.5 and klipper 0.7

I got a bltouch (first i tried 3d touch, but then i got original bltouch 3.1) Both having same issue: during startup pin moves, but klipper says : Failed to verify BLTouch probe is raised; retrying.

And when I'm trying to send DELTA_CALIBRATE it probes only one point and then stop and fails to deploy (BLtouch failed to deploy) or fail to raise (BLtouch failed to raise probe).

I tried to change pin_up_touch_mode_reports_triggered and pin_up_reports_not_triggered from false to true. I tried all 4 different combinations and it fails with all 4 of them. If both false it fails to deploy, if both true it fails to raise probe. I added a resistor between signal and power wire and i also removed a capacitor. 
When i'm trying to run query_probe it always triggered. 

Does anyone here had any luck with bltouch on delta printer? If so, can you share your config?? I have no idea what could be wrong with my config or my hardware

Arhigos, 2019-09-01:
Oh. I'm sorry. i'm fairly new to this. Here is my log file

[klippy.log](https://github.com/KevinOConnor/klipper/files/3563494/klippy.log)

jlirochon, 2019-09-02:
I have a genuine bltouch 3.1 and it works as expected with default settings, however my printer is cartesian.

I think you should install klipper from master branch because the last release is pretty old. At least bltouch 3.1 was released way after klipper 0.7, and if fixes were required to support it, they are not included in 0.7.

I would then ensure the bltouch itself is able to deploy, retract and probe as expected before doing a `DELTA_CALIBRATE` that will probe multiple points. Just issue some `BLTOUCH_DEBUG` commands then try to probe a single point. See [BLTouch](https://www.klipper3d.org/BLTouch.html). If it works on a single point, try `PROBE_ACCURACY` to probe this single point multiple times. Adjust `speed` and `sample_retract_dist` until you are satisfied by the accuracy. If you don't retract enough the pin will hit the bed before it can fully deploy.

When you feel confident, move to more advanced stuff.

Arhigos, 2019-09-03:
I can move the pin using DEBUG command and it works perfectly without any issues. (i can move it many times) It also moves up and down few times when i start my system 

PROBE_ACCURACY doesn't work and it fails after first probe. I tried different speed and sample dist and had no changes at all. I think my problem is that my sensor always report TRIGGERED state, no matter what the pin position is. Maybe because i don't have x,y and z axis? Klipper uses a,b and c instead

Do you have manual how i can upgrade to master branch? I also tried older 3dtouch clone and result was same.

jlirochon, 2019-09-03:
The [installation instructions](https://www.klipper3d.org/Installation.html) are for master branch:

```
git clone https://github.com/KevinOConnor/klipper
./klipper/scripts/install-octopi.sh
```

I know you already did what is specified in the troubleshooting section of [BLTouch](https://www.klipper3d.org/BLTouch.html) page, but in my limited experience the genuine BLTouch v3.1 doesn't require `pin_up_touch_mode_reports_triggered: False`. At least I don't have it in my config when using master branch, and it works as expected. Did you verify that your BLTouch is genuine ?

When I was looking for a way to plug the BLTouch to my Ender 3, I read that removing the capacitor was better that adding a resistor. This is [what antclabs recommends](https://www.antclabs.com/wiring3-1). I don't think you should do both. The pins where I plugged my BLTouch have no capacitor so I did nothing to my board. I followed [this post](https://www.reddit.com/r/ender3/comments/c02q88/the_definitive_guide_to_wiring_bltouch_v3_on_icsp/) for my Ender 3.

Did you try with Marlin ? Maybe you can find a tutorial for BLTouch 3.1 + Marlin that covers your printer specificities. If it works with Marlin, it will be a good foundation and you can come back here to see how you can get Klipper working too.

BlackStump, 2019-09-03:
https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md
https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#how-do-i-upgrade-to-the-latest-software
have you checked to make sure your sensor pin is correct?
https://github.com/KevinOConnor/klipper/blob/master/docs/BLTouch.md

Bltouch is not dependent on the printer kinematics, so has nothing to do with delta's a, b, c which is the towers it is very confusing to label a delta printer towers xyz

Arhigos, 2019-09-08:
Ok. So, i just installed a master branch. I also reinstalled it on new fresh hardware with new fresh OS. So. no old files from previous system. Nothing changed, it still triggered all time. 

As i said, i tried all possible combinations of pin_up_touch_mode, including removing it from config file. I think my Bltouch is genuine, i ordered it from antclabs website (one of the official stores) 

I tried all possible combinations with capacitor and resistor (removing both, installing both, installing only one.)

My sensor pin is correct, because it will not move when you have incorrect pin number, or disconnected white wire. My sensor works, bu it stops working after first probe and always report "triggered"

KevinOConnor, 2019-09-14:
I'm not sure there's further assistance we can provide here.  If you've followed the instructions at https://github.com/KevinOConnor/klipper/blob/master/docs/BLTouch.md and are unable to get a consistent signal from the sensor pin, then that needs to be fixed at the hardware or configuration level.

-Kevin

Arhigos, 2019-09-17:
Do you know how i can "read" or "simulate" output from bltouch sensor? I have all electronic tools - oscilloscope, multimeter, process-calibrators, etc. 

I want to make sure that hardware is working before i go deeper to software troubleshooting.

BlackStump, 2019-09-17:
I would suggest putting that question to the manufacture of the bltouch
ie
https://www.antclabs.com/bltouch-v3

paulsboutique, 2019-09-19:
I had a few issues with my bltouch 3.1, using it on a corexy and would get the same where i could deploy the probe but it would fail at sensing. 
My fix was to power the bltouch with a separate 5v straight from the power supply and added the following to my config 

pin_move_time: 0.4
pin_up_touch_mode_reports_triggered: False

KevinOConnor, 2019-10-08:
I'm going to close this issue as it appears the conversation has concluded.

-Kevin

closed 2019-10-08 by KevinOConnor