Klipper3d/klipper #1932: Macro called Recursively

SrGeek, 2019-09-03:
I got the message macro PRIME called recursively.  The following is my macro
[gcode_macro PRIME]
#default_parameter_EXTRUDER: T0
default_parameter_LENGTH: 15
gcode:
	SAVE_GCODE_STATE NAME=prime_it
	{% if EXTRUDER== T0 %}
	G91 ;relative coordinate
	{EXTRUDER}
	G1 E{LENGTH} F200
	G1 E-1.5 F3000
	{% else %}
	G91 ;relative coordinate
	{EXTRUDER}
	G1 E{LENGTH} F200
	G1 E-1.5 F3000
	{% endif %}
	RESTORE_GCODE_STATE NAME=prime_it
as you can see from my log file.  After I call it again it works.  I get this message everytime I cycle between T0/T1.  Then works after 2nd attempt.  I did remove the "#" from the 1st variable EXTRUDER with the same result.

see klippy.log attached.
[klippy.log](https://github.com/KevinOConnor/klipper/files/3567619/klippy.log)

KevinOConnor, 2019-09-06:
A macro may not invoke itself (either directly or indirectly).  If you call PRIME, which calls T0, which calls PRIME (via its activate_gcode section) then it will result in the given error.

-Kevin

SrGeek, 2019-09-11:
Thanks.  I couldnt figure out where it was calling itself.

closed 2019-09-11 by SrGeek

Aussie84d, 2021-12-05:
So how can I make a Macro Button of, let's say Probe Accuracy?

[gcode_macro PROBE_ACCURACY]
description: Probe test the current location 10 times
gcode:
    PROBE_ACCURACY

That returns :

!! Macro PROBE_ACCURACY called recursively