Klipper3d/klipper #1933: Issues reconnecting

nakaori, 2019-09-03:
Hi, 

first of all: I know about FAQ section 3 and 4 and I already followed these. I still have issues when I try reconnecting to octoprint to my printer after a power cycle.

I'm using an Ender 3.

In Octoprint under Settings -> Serial Connection -> General I configured Serial Port as /tmp/printer with a baud rate of 115200. Additional serial ports:

```
/tmp/printer
/dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0
```

In the Octoklipper settings -> Basic -> Serial Port 
`/dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0`
is set. 

The Klipper config contains 

```
[mcu]
#serial: /dev/ttyUSB0
serial: /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0
[printer]
kinematics: cartesian
max_velocity: 300
max_accel: 3000
max_z_velocity: 5
max_z_accel: 100

```

When I try to reconnect after I turned the printer off and on again the Terminal reading says:

```
Recv: ok B:22.5 /0.0 T0:22.8 /0.0
Send: M105Recv: ok B:22.4 /0.0 T0:22.8 /0.0
Recv: // Klipper state: Shutdown
Connection closed, closing down monitor
Changing monitoring state from "Operational" to "Offline"
Connecting to: /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0
Changing monitoring state from "Offline" to "Opening serial port"
Connected to: Serial<id=0x733a9410, open=True>(port='/dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from "Opening serial port" to "Connecting"
Send: N0 M110 N0*125
There was a timeout while trying to connect to the printerChanging monitoring state from "Connecting" to "Offline"Connection closed, closing down monitor
```

When I restart Octoprint from the commandline or just reboot the whole Raspberry everything returns back to normal.

According to the syslog, the printer indeed changes from /dev/ttyUSB0 to /dev/ttyUSB1. But why does that (apparently) matter after the change of the device path?

[klippy.log](https://github.com/KevinOConnor/klipper/files/3568016/klippy.log)

theopensourcerer, 2019-09-03:
Octoprint *only* needs to know about /tmp/printer, not the device file...

Also, did you follow the installation instructions[1] carefully and complete this step in Octoprint:

> In the Settings tab, navigate to the "Behavior" sub-tab and select the "Cancel any ongoing prints but stay connected to the printer" option. Click "Save".

[1] https://github.com/KevinOConnor/klipper/blob/master/docs/Installation.md#configuring-octoprint-to-use-klipper

nakaori, 2019-09-03:
Thanks for your answer. 

Yes I did that. 

Under additional ports
/dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0 is added automatically. I didn't insert that.

FHeilmann, 2019-09-03:
make sure to disable `modemmanager` on your linux installation as that is known to interfere with the operation of klipper

nakaori, 2019-09-03:
@FHeilmann It isn't installed on my Octoprint image. 

What I found out now is, that if I let Octoprint stay connected to /tmp/printer and just unplug the 3D printer and put it back on, all I have to do is press the restart firmware button in octoklipper to get it to reconnect.

FHeilmann, 2019-09-03:
Glad you got it solved. That makes somewhat sense, the host code and firmware code need to get back on the same page, and restarting the firmware is a good way to accomplish that!

Make sure to close this issue if your problem is solved!

Bests,
Florian

nakaori, 2019-09-03:
I don't know if this is solved. Is this desired behavior? Like is this working like this for everyone?

FHeilmann, 2019-09-03:
A possible fix for that is to restart klipper when the serial port of the 3d printer board comes up again.

See this comment for a guide on how to setup a udev rule for that:
https://github.com/KevinOConnor/klipper/issues/835#issuecomment-441186301

nakaori, 2019-09-03:
Thanks. I will use this as a workaround.

closed 2019-09-03 by nakaori