Klipper3d/klipper #1934: Feature Request: M701 (filament Load) - Stallguard

tiaanv, 2019-09-03:
I thought this might be an interesting one to tackle, albeit a somewhat small group that may find use for it.

My CoreXY style printer has a bowden setup. Loading filament is generally not a hassle, but life could always be easier right?

M701 gcode is the filament load command.  It generally involves calling the command with a distance, optional extruder number, and optional Z lift.  That part is boring, and needs no dev, as it can easily be achieved with normal G1 commands.

Here's the interesting part:  My extruder stepper also has a TMC2209 like all my other steppers, and as such we can make use of the DIAG pin for Stall detection.  What if we could upon loading filament using the M701 command monitor the diag pin, and therefore detect the backpressure when the filament hits the nozzle. 

Filament perfectly loaded to the right position every time.

Normal TMC stallguard threshold settings and virtual endstops would apply, and be used?

I'm sure someone else must have thought about this before, so I doubt I can take credit for the idea.

workheart, 2019-09-05:
would be a interresting idea, thou
I have a TMC2130 for my extruders and use a gcode macro to load fillament, works fine, 
I load the fillament with a G1 in the length of the Bowden plus 100mm in high speed and super low current (150mA) so it dowes not grind the Fillament and loads up to the point i need it, 
i am not sure if stallguard works as the fillament is kinda spongy and you need to have i a rather rapid stop so it would not work if the heater is hot too. 
but you can test it if you would just define your extruder as any other axis and play around with speed currend and stallguard that way there would at least be a proof of concept

theophile, 2019-09-05:
Great idea. It would also be helpful in multi-material/single-extruder setups that rely on scripted unloading and reloading of filament. When my prints fail, it's usually because the prior unloading of the filament left a little "bulb" at the tip which makes it catch on the edge of the throat tube the next time it's loaded. If I hear the clicking and catch it in time, sometimes I can get it back on track by ramming the filament by hand. It would be really nice if the printer could tell when the filament has hit an obstacle before it is "supposed to."

tiaanv, 2019-09-06:
@workheart .  Your approach is also a good one... dial the stepper current down enough to stall effectively...  Your comment on the hard stop is a potential issue...  To test the stallguard theory could be a bit tricky, as you would need some changes to the firmware/software to enable stall detection first.

@theophile. I think potentially this concept can be applied to a number of additional applications on the extruder, provided it actually works in principle.

I actually did some searching now, and came across this:
https://duet3d.dozuki.com/Wiki/Stall_detection_and_sensorless_homing
in the "Using extruder stall detection during filament loading" section.  As I suspected, I am clearly not the first person to think of this, so it might actually be plausible. It would be interesting to know if anyone actually uses this feature in the DUET firmware.

The current irony is that I **temporarily** abandoned Klipper, as I simply could not get my setup:

- SKR PRO 1.1
- TMC2209
- BLTOUCH 3.1
- CUSTOM BUILT COREXY

To work... Specifically, the BLTOUCH behaviour is not right, All of it currently working on Marlin. That's not my end goal, but I need to get the printer operational again. I digress. :)

KevinOConnor, 2019-09-14:
FWIW, I don't know if stallguard would work in this role.  If someone wishes to implement it, test it, and share their results that would be great.

-Kevin

mental405, 2019-09-18:
There is a way to measure the load angle which is more akin to an analog response but that requires polling the registers. It is basically the register that the internal comparator is using to determine if it has exceeded the stallGuard Threshold. 

https://www.youtube.com/watch?v=Prw7wNa20Gk