Klipper3d/klipper #1938: BLTouch 3.1

CappyT, 2019-09-04:
I'm having issues with BLTouch "Smart" v.3.1.
After removing the capacitor on my Z_Endstop (i have a melzi board and AntClabs suggest to remove it) the frequency of bltouch going bad increased dramatically, so much that now it is impossible to complete a 7x7 bed_mesh.

Before the capacitor removal, the bltouch goes bad 1 time in around 1000 probes
After the capacitor removal, the bltouch goes bad 1 time in 25 probes

I think this issue is linked to this one: https://github.com/KevinOConnor/klipper/issues/1882

So I hope this will be fixed soon, as right now there's no way i can use my BLTouch.

Thanks for the help. Attached my printer.cfg (in a zip file) and my klippy.log

[klippy.log](https://github.com/KevinOConnor/klipper/files/3573590/klippy.log)
[printer.zip](https://github.com/KevinOConnor/klipper/files/3573595/printer.zip)

jlirochon, 2019-09-04:
Hi,

Is your printer an Ender3 or something else ?

For the Ender3 I suggest [this post](https://www.reddit.com/r/ender3/comments/c02q88/the_definitive_guide_to_wiring_bltouch_v3_on_icsp/). I tested on mine and my BLTouch 3.1 works so far, although I didn't test with 1000 probes.

CappyT, 2019-09-04:
@jirochon
Yes, my printer is an ender 3. I will try today if it works.

tonn0297, 2019-09-04:
i might add that marlin holds pwm on the probe at all times. and klipper do not. I think that a part of the problem. i had the same issue with mine 3.1, but with marlin it works no problem. i too want this fix so i can use klipper again.

CappyT, 2019-09-05:
> i might add that marlin holds pwm on the probe at all times. and klipper do not. I think that a part of the problem. i had the same issue with mine 3.1, but with marlin it works no problem. i too want this fix so i can use klipper again.

Have you tried klipper with bltouch on icsp also?

mental405, 2019-09-05:
> holds pwm on the probe at all times.

I am not sure what this means.

tiaanv, 2019-09-06:
I have the same issue.  Same setup, but on klipper probing fails at a high rate.  One thing I clearly noticed.... is in with Marlin when the BLTOUCH does a probe, the moment it triggers it jumps to the up position...  With klipper this does not happen.  Not sure if it's just a symptom or an actual underlying issue.

CappyT, 2019-09-06:
> I have the same issue. Same setup. Marlin works 100%, but on klipper probing fails at a high rate. One thing I clearly noticed.... is in with Marlin when the BLTOUCH does a probe, the moment it triggers it jumps to the up position... With klipper this does not happen. Not sure if it's just a symptom or an actual underlying issue.

Noticed that also. With Klipper, the bltouch touches the plate, Z stops, but after a second/second and half retracts the pin.
I think that can be related. Can you attach your klipper logs? that could help.

tiaanv, 2019-09-06:
@CappyT. I had to **temporarily** go back to Marlin to use the printer.  I will play around again this weekend and attach some logs

@jlirochon , I tend to not think this is pin related, as the EXACT same hardware (SKR PRO 1.1) works consistently in Marlin for me.

jlirochon, 2019-09-06:
@tiaanv 

> with Marlin when the BLTOUCH does a probe, the moment it triggers it jumps to the up position... With klipper this does not happen

Not sure I understand what you mean here.

If you have a reproducible case this is great. I think the following could help :
- marlin configuration
- klipper log (configuration is included)
- videos of marlin/klipper behavior ?

tiaanv, 2019-09-06:
@jlirochon 

> Not sure I understand what you mean here.

It's simply my observation that the BLTOUCH behaves differently when doing a probe action in Klipper than in Marlin.  That, and the fact that is doesn't work reliably in Klipper (yet).

As mentioned I will play around again with Klipper, and post as much detail as possible (including your request above). Busy doing some prints at the moment, so no playtime.

tonn0297, 2019-09-06:
i have a new bltouch on the way, ill make some tests tomorow and see if i can get some debug stuff. 
heres a video of the problem: https://vimeo.com/353339123

if you look the blue light that indicates that the probe are getting pwm is only lit when the probe is getting told to do somthing. in marlin it allways gets a signal ie. the blue light is allways on. i think thats the main problem. i cant really tell from the code in marlin how they do it or how i can test it on klipper. if anyone can make an example of that ill be happy to test it.

KevinOConnor, 2019-09-06:
In order for me to assist, I'll need the log from the Klipper log file from the incident, and it will be necessary to issue an M112 within a few seconds of the error arriving.  (@CappyT - your log is helpful, but unfortunately the M112 came a bit too late, if you can get a better log that would be great.)

-Kevin

tiaanv, 2019-09-07:
OK... Here goes

**First a disclaimer.**  This is NOT an effort to prove anything, it is simply an effort to provide as much information as possible.  It is quite possible that the scientific method is failed in these tests.  PLEASE don't rip my findings apart, I am making no conclusions here!!!

**INTRODUCTION**
So I did a number of tests. Attached is a zip file with:
- Marlin Configuration
- Clipper Configuration
- Various Klipper logs (explained later)
- Videos of Marlin and Klipper probe actions

**MARLIN**
This was the first test I did.
Procedure followed:
- G28 Z
- M48 P5

The first video _1. Marlin.mp4_ depicts the probe action.
Both the Homing and the probing were successful.  I did this test multiple times, and also at higher probing speeds.  I could not get it to fail.

Right at the end of my tests I reverted back to Marlin, and did an M48 P20, just to get Marlin to do many probes, and it still had no issue. The is no video for that.

**KLIPPER**
These tests were, unfortunately, a bit more messy, as I had to RESET the BLTOUCH, restart the Klipper service, and try to get useful information.
I did multiple tests, some were successful, some failed on homing the Z-axis, some failed during the probing action. They failed inconsistently. Some at the start, some during, and some at the end.
Procedure followed:
- G28
- G1 Z10 / multiple 1mm Z jogs
- PROBE PROBE_SPEED=2.5 SAMPLES=5 SAMPLE_RETRACT_DIST=5

In some cases I needed
- G1 Z10 (to lift
- BLTOUCH_DEBUG COMMAND=reset

The order of these should be visible in the logs.  

**Videos:**
The first video test was done after homing successfully.  I do a PROBE (failed), then RESET, and try the PROBE again. Video _2. Klipper A - FAIL and SUCCESS.mp4_

The next test I did was a good one.  The probe completed successfully. Video _3. Klipper B - SUCCESS.mp4_

The final video test fails on the second probe action. Video _4. Klipper - SUCCESS and FAIL.mp4_

**Logs**
There are a few logs attached.  The first log is a bunch of tests but has no M112, I only later did tests with M112. They are also attached.

The two main logs of interest:
_2. klippy-Failed G28.log_ depicts a failed homing sequence.
_3. klippy-Probe Failed after 14 good.log_. This is a good one, as I manage to get 14 good probes, then a failure.

I hope all of this makes sense, and can somehow help.  Let me know if I need to explain/clarify something.

[PROBE TEST.zip](https://github.com/KevinOConnor/klipper/files/3586342/PROBE.TEST.zip)

KevinOConnor, 2019-09-07:
@tiaanv - Okay, your log indicates the root of the problem is message retransmits - almost certainly an error in the new STM32F407 USB support.  I'll take a look at it.  It seems unrelated to the original problems report in this issue.

-Kevin

tonn0297, 2019-09-07:
i made some tests. 
the log shows:
G28
BED_MESH_CALIBRATE 
and the probe failed 
M112
i hope this will help.
[klippy.log](https://github.com/KevinOConnor/klipper/files/3586611/klippy.log)
ill swich to the bltouch 3.0 and try the exsat same thing and see if i can get it to fail with that.

tiaanv, 2019-09-07:
> @tiaanv - Okay, your log indicates the root of the problem is message retransmits - almost certainly an error in the new STM32F407 USB support. I'll take a look at it. It seems unrelated to the original problem report in this issue.
> 
Interesting.  Was not anticipating that. Thx Kevin.

tonn0297, 2019-09-08:
And i've talked to one of the dev's fro marlin. he says antslabs if recomending to hold pwm on the probe at all times.
(quote)
we have talked about whether or not the pwm state needed to be held and Paris was quite insistent that it did. I also recently had some discussion with someone who is doing some comparisons between Marlin and klipper who noticed that they shut off the pwm and he found there is no difference in repeatability with it off.

KevinOConnor, 2019-09-08:
@tonn0297 - both of your logs indicate the Klippy host software had some kind of internal race condition - after probing, the host software successfully queried the stepper position from the micro-controller, but for some reason didn't initially recognize the result.  It re-queried a couple of hundred millseconds later, but by that time the bltouch went into an error state.

This error is not similar to the original problem report in this issue.

Unfortunately, I've been unable to reproduce this timing race locally.  If you can easily reproduce the problem, try pulling the test code below, run the probe until failure, run M112 within a few seconds, then attach the log here.

It's important that the test be run with the pristine code - this command will revert your local changes - so if they are important, back them up first.

```
cd ~/klipper ; git fetch ; git checkout origin/work-debug-20190908 ; git reset --hard origin/work-debug-20190908 ; sudo service klipper restart
```

-Kevin

CappyT, 2019-09-09:
> @tonn0297 - both of your logs indicate the Klippy host software had some kind of internal race condition - after probing, the host software successfully queried the stepper position from the [...]

You are a legend. I'm currently returning my ender 3 pro for some other problems with it and i'm getting a i3 Mega-S instead. I will try klipper and bltouch on that, to see if the issue also exists on different printers (Mega-S is based on Trigorilla mainboard, which is RAMPS 1.4 + Mega 2560 in a single PCB).
Then I will also run your test and attach both here ASAP.

Thank you very much for your help.
Also, there's a donate button located somewhere? =D

tonn0297, 2019-09-09:
Thank you Kevin.
Im just about to finish a pro t and then ill swich to klipper and get some logs.
Thank you for your help. 
Shuld i make a new problem so this thread is not filled?

tonn0297, 2019-09-09:
@KevinOConnor 
heres a new log with the Git reset and all that

[klippy.log](https://github.com/KevinOConnor/klipper/files/3591055/klippy.log)

just to be sure here's my setup hardware.
SKR V1.3
watterrott TMC 5160 spi 32mS
Raspi 4 4gb 
octoprint v1.3.11
octopi v0.17.0
BlTouch v3.1 and 3.0 Genuine
24V psu

KevinOConnor, 2019-09-09:
@tonn0297 - looks like a Klipper bug.  Can you retest with the latest code on that branch to see if the problem goes away?

cd ~/klipper ; git fetch ; git checkout origin/work-debug-20190908 ; sudo service klipper restart

-Kevin

tonn0297, 2019-09-09:
@KevinOConnor you are the man... 
ill make some more tests now just to make shure. 
what was the change?

tonn0297, 2019-09-09:
So i just did 2400 probe no problem. so i think we are good. 
again thank you kevin for all your help and you input to the community.
-Tonny

KevinOConnor, 2019-09-10:
@tonn0297 - Okay, thanks.  If you still have the log from the 2400 probe attempt, could you attach it here?  (If not, then don't worry about it.)  Feel free to use the workaround on the work-debug-20190908 branch.  Unfortunately, it's not in a shape that can be committed to the master branch - I'll have to think about how to handle the underlying issue.

In summary, I'm aware of the following issues:
1 - The stm32f407 had a usb stability issue (should be fixed as of 4fa41d9c).
2 - There is a race condition on host to mcu query requests that could cause query responses to get delayed ( https://github.com/KevinOConnor/klipper/issues/1938#issuecomment-529255057 ).  That delay could cause the bltouch to go into an error state.  Work around on work-debug-20190908 branch.
3 - A low level retransmit could cause a delay in processing a query request. (Part of https://github.com/KevinOConnor/klipper/issues/1938#issuecomment-529101191 .)  This delay also results in the bltouch going into an error state.
4 - The code attempts to clear the error state from the bltouch, however it can't clear the error state when the bltouch is close to the bed.  Ideally, the code would raise the probe before trying to clear the error state.

-Kevin

tonn0297, 2019-09-10:
@KevinOConnor 
sadly i dont have the log. but i can do a 20x20x3 probe when i get home and send you the log. 
And i can see i have some problems with the tmc5160 extruder resetting to default 256 microsteps and so on. but ill make a new issue when i know more.
Kind regards Tonny.

jlirochon, 2019-09-11:
Hi @KevinOConnor I did a 40 x 45 x 3 MESH_BED_CALIBRATE, and at some point I got the error `!! BLTouch failed to raise probe`. I'm on **work-debug-20190908** branch. My board is a creality3d 1.1.4 (ender3) and I have a genuine bltouch 3.1. I issued a `M112` a few seconds after the failed probe.

[klippy.log](https://github.com/KevinOConnor/klipper/files/3601126/klippy.log)

tonn0297, 2019-09-11:
i think they says it has a failure rate of 5%.
i just did a 20x20x3 again and i have no problem. heres a log 
[klippy.log](https://github.com/KevinOConnor/klipper/files/3601652/klippy.log)

jlirochon, 2019-09-13:
@KevinOConnor thanks

The bed on the ender3 is cheap and warped, I was trying to probe it using a 5mm grid to see how bad it looks, but it always fail at some point. I tried work-debug-20190908 for this reason, after reviewing the changes you did on it. I must admit I don't know if it fall specifically under the race condition problem. I just know I can't complete a BED_MESH_CALIBRATE because it always fail at some point.

I have no prior experience with bltouch sensors, and don't know if it would work better using another firmware or if bltouch are just crap. I can try to reinstall marlin and see if it does better or not. I can try with another version of the bltouch too (I pinched an older one from an older printer).

> Otherwise, item 4 still applies

I have some python skills and I'm glad to help. As your code is unfamiliar to me, any advice would be of great help.

nophead, 2019-09-13:
I see the previous comments about Marlin doing constant PWM and klipper not. When klipper stops PWM does it ensure the last pulse is clean? If not I can imagine the BL touch acting on the runt pulse.

KevinOConnor, 2019-09-13:
@jlirochon - Ah, I thought your bltouch was working (as in https://github.com/KevinOConnor/klipper/issues/1938#issuecomment-527993276 )?  FWIW, your log appears similar to the log of the original report in this issue.  Alas, both logs didn't get the M112 fast enough, but when I back tracked through the timing I didn't see any issue.  It appeared Klipper responded as designed.

One thing you could test would be to reduce MIN_CMD_TIME in bltouch.py to `2 * SIGNAL_PERIOD`.  (Be sure to run `sudo service klipper restart` after any code changes.)  It may be that the bltouch isn't happy with a 100ms turnaround time in some rare cases.

-Kevin

KevinOConnor, 2019-09-13:
Hi @nophead,

I haven't seen an indication that there is an issue with the PWM signals.  If the bltouch didn't get the signal then I'd expect to get reports of head crashes.  All the logs and reports that I've seen indicate the bed stopped moving and therefore the bltouch successfully signalled the Klipper mcu code and the mcu code successfully stopped issuing step commands.

To answer your question though, the Klipper timing should be precise - exactly 5 pin_down commands over 100ms.

FWIW, I don't think we can leave the PWM signal enabled during probing.  There were indications that leaving the PWM signal on was causing head crashes (and other errors) on some of the clone devices.  Specifically, there was a concern that some clones would not detect a bed touch if the touch raced with a repeat reception of the "pin_down" command.  Similarly there was a concern that some clones would become non-operable if they received a repeat reception of a "pin_down" command immediately after a bed touch event.  Finally, in order to issue the pin_up command after a bed touch (on clones or original bltouchs), we'd need to ensure the pin_up command does not overlap with a repeat "pin_down" command - getting that timing correct is complex and would ultimately lead to further delays of the "pin_up" command.

-Kevin

KevinOConnor, 2019-09-13:
@jlirochon 

> > Otherwise, item 4 still applies
> 
> I have some python skills and I'm glad to help. As your code is unfamiliar to me, any advice would be of great help.

For item 4, my thinking is that it would require some changes to the high-level probing procedure.  Right now, a probe event (as implemented in probe.py) involves moving the head down until receiving an "endstop" signal.  In order to better handle the bltouch, this may need to change to something like: move down until trigger and then move up.  In this way, probe.py could signal bltouch.py after the move up event, and bltouch.py could perform the reset request (if needed) after the head has moved up.  (Right now, the bltouch.py code attempts to do the reset immediately after the bed touch event - while the head is still near the bed.)

Also, be aware of the general developer documentation at: https://www.klipper3d.org/Overview.html#developer-documentation

-Kevin

epandi, 2019-09-25:
I hope it's the same  problem as the rest describes, but I'm facing problems that sometimes the probe is being not pulled up after touching the bed, which results in the bltouch (v3.1) is going into alarm state. Im running the debug 0908 branch of klipper. 
[klippy.log](https://github.com/KevinOConnor/klipper/files/3651596/klippy.log)

jlirochon, 2019-09-25:
Hi @epandi,

**I think I have found a workaround to this problem**. At leat I fixed the problem for my printer, which is an ender3 with original creality3d 1.1.4 board.

My branch need some cleanup because I don't want to introduce bugs for other versions of BLTouch. **If you are interested in testing this branch let me know**.

What I did:
* keeping PMW signal during the probe, blue led stays on until the pin touches the bed (the way it should be, as stated in antclabs documentation) => no effect
* slightly adjusting frequencies, because they changed on v3.1 => no effect (as I expected because they have a tolerance margin of 20% anyway)
* probing with *Touch Switch Mode* mode instead of *Push-Pin Down Mode* => **works for me !**

Not sure what's the difference exactly. According to antclabs documentation the probing sequence is similar, except for *Push-Pin Down Mode* they say "Caution: An alarm may occur". Klipper fails to clear the alarm but as it appears randomly it's difficult to debug. I'm waiting for more details from antclabs to know if *Touch Switch Mode* has some downsides. This mode is available from BLTouch Smart 2.1 up to current versions.

Honeybaadger, 2019-09-25:
> Hi @epandi,
>
> **I think I have found a workaround to this problem**. At leat I fixed the problem for my printer, which is an ender3 with original creality3d 1.1.4 board.
>
> My branch need some cleanup [...]

im interested in getting my ender3 w/ BLtouch running :) how can i test it

jlirochon, 2019-09-26:
@epandi @Honeybaadger @nophead @tonn0297 @CappyT 

I have [pushed my work in progress](https://github.com/jlirochon/klipper/tree/feature/bltouch-improvements). Feedback is welcome, I would like to know if it works for you ! Feedback on the code itself may wait until I submit a pull request.

Quick instructions for those non familiar with git:
```
git remote add jlirochon https://github.com/jlirochon/klipper.git
git checkout -t jlirochon/bltouch-improvements
```

Here is an example from my `printer.cfg`:
```
[bltouch]
flavor: genuine_smart_3.1
sensor_pin: ...
control_pin: ...
x_offset: ...
y_offset: ...
speed: 2
samples: 3
samples_result: median
sample_retract_dist: 5
```
**`flavor=xxxxxxxx` tries to introduce sensible defaults for each flavor. You have to set a flavor in order to enable my patch** (no flavor = backward compatible defaults = current behavior). You can still override as usual, for example you can set `pin_up_touch_mode_reports_triggered: False` in your config, but in my experience there is no need to change this for the genuine bltouch 3.1. If you have desperately tried settings you don't understand and observed no benefit, please remove them prior to testing, start with a minimalist `[bltouch]` config and see how it goes.

epandi, 2019-09-27:
@jlirochon For me this change was a game breaker. I don't have any problems using the bltouch 3.1 anymore. All attempts bed levelling (around 20 3x3 bed meshed) succeded without any problems.

HaythamB, 2019-09-27:
@jlirochon Your branch solved the issues for me. Anycubic Mega, running TMC2208 (no UART), Trigorilla board. Sensor pin connected to ar19.
The master klipper branch just wouldn't take the 3.1 Smart no matter what. I think your changes are close to what is the best solution at the moment. It still sometimes fails, and I still must switch both trigger options to False for it to work properly. But it works.

jlirochon, 2019-09-28:
@HaythamB I appreciate your feedback, thank you !

> It still sometimes fails, and I still must switch both trigger options to False for it to work properly. But it works.

Interesting... I would like to understand why 2 instances of the same BLTouch would require different parameters. This challenges the concept of flavor I want to introduce.

* Are you sure you have a genuine BLTouch 3.1 from antclabs (did you verify the qrcode ?)
* What's the exact value of your `sensor_pin` in `[bltouch]` section ?
* Do you have more information about your Trigorilla board ? Version of the board ?
* How did you physically connect the sensor ? Did you remove a capacitor from the Trigorilla ? Set a pull-up resistor ? Did you set the output voltage of the BLTouch, or is it operating in open drain mode (default) ?

I hope we can diagnose a problem and make it work with consistently with the default parameters.

HaythamB, 2019-09-28:
@jlirochon it is puzzling indeed. I ran 3x3, 5x5, and 7x7 validations, and they tended to fail on the 5th or 6th probe. Its definitely not consistent. I have a feeling it has to do with the timings of the PWM in klipper, since the raspberry pi is not a real-time OS and may be causing some sort of mcu timeout?

Here's what you asked for:
- Its a genuine 3.1, I have verified the serial number.
- ar19 (config below)
- Its the 8-bit board with the removable drivers. [Here's a link to a similar Amazon listing.](https://www.amazon.co.uk/ANYCUBIC-TriGorilla-Integrated-Compatible-Controller/dp/B0788HM2CP)
- The sensor is connected through the filament sensor harness, which is passed through a PCB extension board inside the printer, and then directly to the Z+ pin and GND pins on the controller board. There are no capacitors/resistors in the signal path from what I can tell. But this brings the length of the wires from the sensor to the pin to approx 160cm. I left all the BLTouch defaults in place, no altering of the logic levels or anything.

Here's the BLTouch section from my config file:
```
[bltouch]
flavor: genuine_smart_3.1
sensor_pin: ar19
control_pin: PB5
#pin_move_time: 0.4
pin_up_reports_not_triggered: False
pin_up_touch_mode_reports_triggered: False
x_offset: 44
y_offset: -8
z_offset: -0.260000
speed: 2
samples: 3
```

BlackStump, 2019-09-28:
@HaythamB 
>z_offset: -0.260000
is that correct? my offsets are 1.1
-0.26 is having the nozzle below bed height 

>It's a good idea to verify that the Z offset is close to 1mm. If not, then you probably want to move >the probe up or down to fix this. You want it to trigger well before the nozzle hits the bed,

HaythamB, 2019-09-28:
@BlackStump yeah I've been having issues with my mount, its not the best. Had to add a 3rd M3 nut to get it close, now the probe (when retracted) is about 0.4mm higher than the nozzle tip. I need some washers instead to increase this gap. Thank you for pointing this out.

@jlirochon This is the reason I was still getting timeouts, the probe needed more room to be triggered. Seems like its working perfectly now. Thanks for the great work!

jlirochon, 2019-09-28:
@HaythamB 

> Seems like its working perfectly now

Great ! were you able to remove `pin_up_reports_not_triggered` and `pin_up_touch_mode_reports_triggered` ?

HaythamB, 2019-09-29:
@jlirochon so haven't been able to get it to work properly. Modified the offsets correctly, but the timing is still off.

Flashed in Marlin 2. Works perfectly out of the box with the same offsets. Standard 3.1 settings there (such as 5V logic level) on the exact same pins I had in klipper. Interruptable endstops option is not enabled either, just in case.

From what I can tell, when the probe starts to hit the bed, it lights up red (detecting the bed correctly), but there's approx a 2 sec delay until the sensor pin is detected in klipper, during which time the pin travels quite a lot into the probe (based on our Z speed, ofc). In Marlin, the pin triggers the sensor almost instantly (like any YouTube video of a BLTouch running normally would show).

Seems to me that there's still a timing detection issue happening in klipper. Leaving Marlin flashed for the time being until we can figure this out, cause I really want to try out the BLTouch! :)

Guess there's still more work ahead. Your solution is about 90% of the way there. There must be come timings to tighten up to have it nailed.

epandi, 2019-10-06:
Just to give an update. I still don't have any issues after running @jlirochon proposed fix. So for my problem (sometimes the pin is not pulled up after the probe touching the bed) it's rock solid.

KevinOConnor, 2019-10-08:
FYI, as far as I know, touch_mode has been present since the very first bltouch.

-Kevin

jlirochon, 2019-10-08:
I'm relatively new to 3D printing and I don't have the history, but the page at [https://www.antclabs.com/bltouch](https://www.antclabs.com/bltouch) says **Probing with Touch Switch Mode(M280 P0 S60)    ← only Smart V2.1 and later**. Maybe they introduced it earlier and don't assume because it didn't work well, I don't know.

CappyT, 2019-10-11:
@jlirochon Can you submit a pull request with your patch?

pusewicz, 2019-10-13:
@KevinOConnor https://github.com/MarlinFirmware/Marlin/pull/15547 Would this be of any use for Klipper?

skrewz, 2019-10-20:
Just a +1. @jlirochon's `flavor: genuine_smart_3.1` made my AntClabs BLtouch 3.1 "smart" probe function.

CodeChief, 2019-10-27:
@jlirochon did you make a pull request? I couldn't find it. Or @KevinOConnor if he doesn't submit one please just copy the bltouch.py over your /klippy/extras/bltouch.py to complete the BLTouch support.  It's really necessary for two good reasons:
1. You hard-code the "none" send command during homing setup but the >2.1 versions require "touch_mode" here.
2. Versions >3.0 require completely different PWM values for many commands making Klipper in it's current state incompatible with BLTouch as sold.
The only minor change I'd suggest during the pull would be to consider rename the new "flavor" configuration setting to "model" or "version" as it's specifying a hardware model/version not icecream.  ;-)

ade, 2019-11-08:
BLTouch smart 3.1 working mostly for me with 0.8 offical release, however, 1 in 10 times or so it will report TRIGGERED when pin is up, but mostly not. This seems to cause klipper to fail to verify that pin is up. Is that related to this issue? Perhaps my probe is faulty?

CappyT, 2019-11-08:
> BLTouch smart 3.1 working mostly for me with 0.8 offical release, however, 1 in 10 times or so it will report TRIGGERED when pin is up, but mostly not. This seems to cause klipper to fail to verify that pin is up. Is that related to this issue? Perhaps my probe is faulty?

It is indeed related. I also had that problem.
Use this branch, since even 0.8 klipper doesn't fix this problem.

ade, 2019-11-10:
Sorry, my issue was a faulty cable, BLTouch smart 3.1 works without issues for me now running release 0.8 on ramps 1.6 with mega2560

CodeChief, 2019-11-10:
Good for you @ade but before the last comment is used to assume this issue is closed again (WRONG!) @KevinOConnor more importantly other people still need at least the single file "bltouch.py" from the @jlirochon pull request.

And @ade did you read about the PWM values being off in the later versions? Go check-out the spec sheet from BLTouch and the code changes to support them then you will see. Even current 3.1 users and later should have this update to ensure support both normally (probing) and for calibration/debugging.

For people who need a solution in the meantime, this is how you get a working BLTouch from @jlirochon with the rest of the current Klipper source code from @KevinOConnor:

```
# Download.
rm ~/klipper --recursive --force
git clone https://github.com/KevinOConnor/klipper.git

# Patch.
wget https://raw.githubusercontent.com/jlirochon/klipper/feature/bltouch-improvements/klippy/extras/bltouch.py -O ~/klipper/klippy/extras/bltouch.py

# Build.
cd ~/klipper/
make clean
make

# Install.
sudo service klipper stop
make flash FLASH_DEVICE=/dev/serial/by-id/[*** YOUR DEVICE ID - SEE DOCUMENTATION ***]
sudo service klipper start

cd ~
```

Hope that helps, would be good to have this closed for good by simply allowing the pull request through.

ade, 2019-11-10:
Sorry, didn't mean to disrupt your conversation, nor deny your problems. I simply stated that it is working for me.

I also merged the fork linked on top of the 0.8 release version and tried it (it works fine, too).
If you want to use that instead, all the names are the same just change the username to ade.
https://github.com/ade/klipper/tree/feature/bltouch-improvements

Lastly, I don't think @jlirochon ever created a PR for this, so perhaps someone else needs to take over it.

jlirochon, 2019-11-12:
Hello,

I rebased my work on top of latest changes from master, and submitted a PR in #2147 

I holded this PR because I wanted to do more. As I didn't have the time yet, It should probably be merged. It would solve issues for some of you and it would avoid dealing with multiple repositories/patching.

KevinOConnor, 2020-02-24:
FYI, with the merge of PR #2532 I believe the Klipper timing issues discussed above should now be fixed.

I am aware, however, of the following outstanding issues with the current Klipper BL-Touch code:
1. Should the bltouch go into an error state the Klipper reset command is not given enough time. The current code will issue a reset command for 100ms before issuing a pin_up command and this time should be longer (see https://github.com/KevinOConnor/klipper/issues/2267#issuecomment-567044957 ).
2. Should the bltouch go into an error state the code attempts to clear the error state from the bltouch, however it may not be able to clear the error state when the bltouch is near the bed. Ideally, the code would raise the probe before trying to clear the error state.
3. There are reports that BLTouch v3 users are having better success with probing in "touch mode" instead of "pin_down mode".  (However, BLTouch v2 and many clones are known to work poorly when homing in "touch mode".)

-Kevin

greyhound3, 2020-02-27:
I had a BLTouch Smart v3.1 working on a Robin Nano v1.2
After switching to a Bigtreetech SKR v1.4 pin could be moved up and down but probing resulted in errors. Thanks to this thread and the "flavor: genuine_smart_3.1" config option it works now!!!
Thanks !!

[5 more replies]

closed 2020-04-24 by KevinOConnor