Klipper3d/klipper #1940: printer.<heater>.target not for extruder hotend?

linux-paul, 2019-09-05:
Hi, I'd like to write a GCode Macro which fetches the extrruders target temperature. After I took a look to the docs it seems that only heater_bed and heater_generic configurations are supported and on the other hand I can't change the extruder heater to a generic one. Is there any solution?

KevinOConnor, 2019-09-06:
An extruder is considered a "heater", so one should be able to use `{printer.extruder.target}`.

-Kevin

linux-paul, 2019-09-06:
Hello Kevin,
this seems the point:

[gcode_macro wait_heater]
gcode = M109 S{printer.extruder.target}

results in:
Error evaluating 'gcode_macro wait_heater:gcode': UndefinedError: 'extras.gcode_macro.GetStatusWrapper object' has no attribute 'extruder'
(or I did something stupid wrong ...)

Tailored klippy.log attached
[klippy_short.log](https://github.com/KevinOConnor/klipper/files/3584797/klippy_short.log)

many thanks,
linuxpaul

KevinOConnor, 2019-09-06:
Ah, the internal name is actually called `extruder0`.  This is something that should be fixed in Klipper.

-Kevin

linux-paul, 2019-09-06:
And maybe also the documentation? command templates show:
- `printer.<heater>.temperature`: The last reported temperature (in
  Celsius as a float) for the given heater. Available heaters are:
  `heater_bed` and `heater_generic <config_name>`.

printer.extruder0.target works perfect.

Many thanks.
linuxpaul

KevinOConnor, 2019-11-07:
FYI, it's now `{printer.extruder.temperature}` as of commit ed610a66.

-Kevin

closed 2019-11-14 by KevinOConnor