Klipper3d/klipper #1942: [FR] fork the octoklipper-plugin into Klipper

Hywelmartin, 2019-09-06:
So @mmone  created a plugin to Octoprint.. but as it seems, it is more or less dead.

There are some bugs that are over a year old.. that is more or less fixed.. the code changes have been committed but not merged..

For newcomers this plugin is a great way to interact with there printer and get a more refined experience of the FW.. and even might render less issues..
One preset per printer type. cartesian, delta, corexy and so on would be even better..

An alternative could be to create a totally new plugin..

@KevinOConnor if you think this out of the scope for Klipper please close the ticket..

KevinOConnor, 2019-09-14:
FYI, maintaining an OctoPrint plugin is not something I personally am ready to take on.

I understand the Klipper OctoPrint plugin is very useful and I'm glad it was made available.  I'm also sorry if it is no longer being maintained.  However, it will require someone to step forward to fill that role.

-Kevin

KevinOConnor, 2019-09-29:
I'm going to close this issue as it looks like the conversation has concluded.  If someone does wish to step forward to maintain a plugin, then let us know.

-Kevin

closed 2019-09-29 by KevinOConnor