Klipper3d/klipper #1943: Pressure advance makes extruder stop below 0.5

TheFoxBurrow, 2019-09-08:
I have the issue that trying to get pressure advance working with values <0.5 makes the extruder stop after the first two walls of the first layer. There is no clicking, no missed steps, it just completely doesn't move anymore. Values higher than 0.5 seem to work alright, however that is way to much for my setup.
I'm running an Ender 3 with the Atmega 1284p version 1.1.5 board from creality with TMC2208s, obviously Octoprint with Klipper, and the standard extruder connected by capricorn bowden to a genuine e3d v6.

I have no idea as to why this is happening. The only other issue i have is the part cooling fan not really being controllable and running at full speed already at 1%. If this seems unrelated, just ignore it, I might open a separate issue for it at some point, just want to state all symptoms right now. I have two 12V 5015 blowers connected with a buck converter to get the 24V down to 12V, 1% software fan power already equals what feels like 100% hardware fan power. Worked fine and as expected with Marlin before installing Klipper, so there should be no inherent electrical issue.

Right now I see two possibilities:
1. My install of Klipper is botched. It for some reason didn't properly work the first time I tried to install, so I formatted the SD card and started anew once already. Second (currently active) install went without hitches, but who knows...
2. I found some kind of bug. As a programmer myself, it is weird to work for 0.5 and up, but absolutely not below.

Let me know if you need further information or some other stuff, thanks!

BlackStump, 2019-09-08:
maybe related to this
https://www.klipper3d.org/FAQ.html#my-tmc-motor-driver-turns-off-in-the-middle-of-a-print

https://github.com/KevinOConnor/klipper/issues/1829

Hywelmartin, 2019-09-08:
The fan issue could be electrical, have a look at this [Video](https://youtu.be/k9Yy8OxohGI)

TheFoxBurrow, 2019-09-08:
So to sum up the findings you presented:
There is no way for pressure advance to work without any hardware modifications on my board with the integrated drivers, since there are no UART pins available. I'm just sol, would have to get a different board? Would adding a geared extruder like a Titan help in any way?
For the fans, I should:
- Connect negative input on the buck converter to power supply ground
- Connect positive input on the buck converter to mainboard fan positive output (as it is right now)
- Connect positive output on the buck converter to the positive wires of the fans (right now wired in parallel)
- Connect the negative wire of the fans to the mainboard fan negative output.
Did I undestand that correctly, @Hywelmartin ? Could i potentially avoid that whole mess by just wiring the two 12V 5015s in series and directly connect them to the mainboard fan outputs?

Thank you for all your help

Hywelmartin, 2019-09-09:
@TheFoxBurrow  It sounds correct..
It may work to put them in series, but I recommend to use the buck converter.. and in parallel

KevinOConnor, 2019-09-14:
> There is no way for pressure advance to work without any hardware modifications on my board with the integrated drivers, since there are no UART pins available.

FYI, it is not valid to use tmc2208s in "standalone mode" on any stepper with Klipper.  They work fine in uart mode, but random shutdowns can occur at any time on any axis when they are in standalone mode (details at https://www.klipper3d.org/FAQ.html#my-tmc-motor-driver-turns-off-in-the-middle-of-a-print ).

-Kevin

TheFoxBurrow, 2019-09-17:
Yes, @KevinOConnor , there is just sadly nothing I can do about it on the creality board I have since the stepper drivers are integrated in the board, no pins available. Or am I missing something here?
My next step would be to get a board like the MKS Gen L and 2208s separately (or would a switch to TMC2209s make sense here?), but since I've already used up my monthly budget for upgrades that'll have to wait.
@Hywelmartin I finally got to switching up the wiring as described above, parallel with a buck converter. Fans work flawlessly now. Should've done that much sooner. Thanks again for pointing me in the direction of Tom's video. I'm one of his subscribers and saw that video when it came out, but since my fan control worked back then, with the old board, I never thought much of it and forgot.

KevinOConnor, 2019-10-08:
I'm going to close this issue as it appears the conversation has concluded.  The only solution for the tmc2208 issue is to configure it in uart mode - if that's not possible then there is no current solution.

-Kevin

closed 2019-10-08 by KevinOConnor