Klipper3d/klipper #1947: REQUEST if possible

bruce356, 2019-09-09:
Hi Kevin, on my IDEX printer I would like to be able to turn OFF the stepper motor to the X2 axis if not being used (so I can position it manually  and utilise the second parts cooling fan for small parts cooling from both sides). Obviously all stepper motors can be disabled / enabled but would it be possible to disable /enable  individual stepper motors while printing or when not printing. - Thanks

tonn0297, 2019-09-09:
looks like its not possible or about to implement.
https://github.com/KevinOConnor/klipper/issues/906

KevinOConnor, 2019-09-14:
As indicated in #906, someone would need to implement and submit that.  If one is using Trinamic stepper drivers (or possibly other mechanisms for run-time current control) then an alternative solution may be to reduce current to the driver when it is not in use.

-Kevin

bruce356, 2019-09-16:
@KevinOConnor Thanks for the reply, yes I am using Trinamic stepper drivers, I had forgotten about #906. If anyone is capable and willing of figuring out the code required it would be a great addition to Klipper and of course I would be very great full. 
Thanks in advance and regards - bruce

JohnEdwa, 2019-12-20:
This is something that I would really like to see added soon, as the lack of this just damaged my build plate.  
 
During testing with the new BLtouch code, I had a failure to deploy the probe which caused Klipper to error out and disable the stepper motors. This caused my gantry to plummet to the bed and now my buildtak has a nozzle-shaped molten dimple on it :/

ld3300, 2020-02-03:
#2463

KevinOConnor, 2020-05-24:
FYI, this was implemented in #2463. 

-Kevin