Klipper3d/klipper #1948: Stepper always on

HodenKlaus, 2019-09-09:
Setup:
Skr.1.3
Tmc2130 with SPI all axis

Didnt attach a log because i dont think it would be useful but can ofc.
One of my Z steppers ist always on. Cant figure out why.

[printer_blv.zip](https://github.com/KevinOConnor/klipper/files/3592503/printer_blv.zip)

HodenKlaus, 2019-09-10:
[klippy.zip](https://github.com/KevinOConnor/klipper/files/3596850/klippy.zip)
is this right file ?

KevinOConnor, 2019-09-14:
You should follow the steps described at https://www.klipper3d.org/Config_checks.html to identify and correct these types of configuration errors.

-Kevin

HodenKlaus, 2019-09-16:
Ofc I did all the mentioned above. Printer runs fine so far. But one z motor is always on.

KevinOConnor, 2019-09-21:
Alas, the only advice we have is listed at https://www.klipper3d.org/Config_checks.html .

-Kevin

KevinOConnor, 2019-10-08:
I'm going to close this as it doesn't appear anyone has an answer for the original question.

-Kevin

closed 2019-10-08 by KevinOConnor