Klipper3d/klipper #1950: Can't figure out how or where to change Stepper E-Steps values

NickTom88, 2019-09-10:
After Finishing the updates I printed a calibration cube and my X and Y values are spot on but Z is about 19.6. Traditionally in Marlin I could change the E-Steps right on the controls or check my settings with M503 and adjust using M92. With Klipper I'm not able to do any of that. I also wasn't able to find ant values when searching the Config file. I looked through the FAQ and support docs and couldn't find anything that addressed this. Can anyone walk me through doing this?

BlackStump, 2019-09-10:
https://github.com/KevinOConnor/klipper/blob/master/config/example.cfg#L36
https://github.com/KevinOConnor/klipper/blob/master/config/example.cfg#L75
https://github.com/KevinOConnor/klipper/blob/master/config/example.cfg#L87
https://github.com/KevinOConnor/klipper/blob/master/config/example.cfg#L101

https://github.com/KevinOConnor/klipper/blob/master/docs/Installation.md#configuring-klipper

NickTom88, 2019-09-10:
To automated response - will try to attached justifiable log.

BlackStump - I see how Klipper identifies steps now. My problem is figuring out the formula to adjust the value. Any How-To I find just helps for marlin type values (X e-steps = 80.00) not now I'm trying to take a Z value of .0025 and adjust it for adjusting a 20mm cube where z is 19.6 but s/b 20. I'm lost on the formula. I don't see any conversations in any of those links you supplied and couldn't good one.

BlackStump, 2019-09-10:
I think it would be
current step distance  multiplied by actual measurement divided by wanted measurement

theopensourcerer, 2019-09-10:
@NickTom88 Try this for calibration. A 20mm cube is pretty useless IMHO.

https://www.thingiverse.com/thing:2484766

NickTom88, 2019-09-10:
BlackStump - that actually seems possible. Though it only changes the value from . 0025 to . 002551. I'll have to print it again when I get home from work with that value.

Theopensourcerer - I'm sure that would make x and y more accurate, but what about z? Which is my main issue. I mean if I wanted higher values I could just doubly the size of the cube to 40mm in theory.

theopensourcerer, 2019-09-10:
@NickTom88 There is a calibration design for Z there too.

NickTom88, 2019-09-10:
@theopensourcerer ahh I see it all now. As for the spreadsheet, how do I co about using the Klipper value? .0025 instead of 400.

theopensourcerer, 2019-09-11:
To switch between marlin style step/mm to Klipper's mm/step simply:

1/klipper = marlin
1/marlin = klipper

NickTom88, 2019-09-11:
@theopensourcerer I feel dumb for not realizing that. I could hug you.

closed 2019-09-11 by NickTom88