Klipper3d/klipper #1952: TMC5160 Extruder resets

tonn0297, 2019-09-10:
i have a problem with my tmc5160 extruder.
it starts the print fine but after a while it stops and dump tmc shows some weird stuff.

before error:
[image]

after it stops:
dump_tmc stepper=extruder
Recv: // ========== Write-only registers ==========
Recv: // TPWMTHRS:   00000002 TPWMTHRS=2
Recv: // COOLCONF:   00020000 sgt=2
Recv: // IHOLD_IRUN: 00060404 IHOLD=4 IRUN=4 IHOLDDELAY=6
Recv: // TPOWERDOWN: 0000000a TPOWERDOWN=10
Recv: // ========== Queried registers ==========
Recv: // GCONF:      00000004 en_pwm_mode=1
Recv: // CHOPCONF:   10410150 hstrt=5 hend=2 tbl=2 tpfd=4 MRES=0(256usteps) intpol=1
Recv: // GSTAT:      00000001 reset=1(reset)
Recv: // DRV_STATUS: 001f4000 stealth=1 CSACTUAL=31
Recv: // FACTORY_CONF: 0000000c FACTORY_CONF=12
Recv: // IOIN:       30000040 SD_MODE=1 VERSION=0x30
Recv: // LOST_STEPS: 00000000
Recv: // MSCNT:      000000b8 MSCNT=184
Recv: // MSCURACT:   006700e0 CUR_A=224 CUR_B=103
Recv: // OTP_READ:   0000000c OTP_FCLKTRIM=12
Recv: // PWMCONF:    00000000
Recv: // PWM_SCALE:  00000080 PWM_SCALE_SUM=128
Recv: // PWM_AUTO:   00490080 PWM_OFS_AUTO=128 PWM_GRAD_AUTO=73
Recv: // TSTEP:      00008aca TSTEP=35530

ill get you an klippy.log when i mange to do a M112 when it happens.

tonn0297, 2019-09-10:
it seems like it the same as https://github.com/KevinOConnor/klipper/issues/1797 and https://github.com/KevinOConnor/klipper/issues/1774

but i dont run pa.

tonn0297, 2019-09-10:
@KevinOConnor 
hey man. heres a log. its random where it stops. the motor is not hot and the drivers has cooling so thats not the issue
[klippy.log](https://github.com/KevinOConnor/klipper/files/3596191/klippy.log)

if there's anything else you need me to test just ping me.

heres a dump_tmc before i started the print:
Send: dump_tmc stepper=extruder
Recv: // ========== Write-only registers ==========
Recv: // TPWMTHRS:   00000002 TPWMTHRS=2
Recv: // COOLCONF:   00020000 sgt=2
Recv: // IHOLD_IRUN: 00060404 IHOLD=4 IRUN=4 IHOLDDELAY=6
Recv: // TPOWERDOWN: 0000000a TPOWERDOWN=10
Recv: // ========== Queried registers ==========
Recv: // GCONF:      00000004 en_pwm_mode=1
Recv: // CHOPCONF:   134141b3 toff=3 hstrt=3 hend=3 chm=1 tbl=2 tpfd=4 MRES=3(32usteps) intpol=1
Recv: // GSTAT:      00000005 reset=1(reset) uv_cp=1(Undervoltage!)
Recv: // DRV_STATUS: 80044000 stealth=1 CSACTUAL=4 stst=1
Recv: // FACTORY_CONF: 0000000c FACTORY_CONF=12
Recv: // IOIN:       30000050 DRV_ENN=1 SD_MODE=1 VERSION=0x30
Recv: // LOST_STEPS: 00000000
Recv: // MSCNT:      000001ec MSCNT=492
Recv: // MSCURACT:   010a001d CUR_A=29 CUR_B=-246
Recv: // OTP_READ:   0000000c OTP_FCLKTRIM=12
Recv: // PWMCONF:    00000000
Recv: // PWM_SCALE:  00000014 PWM_SCALE_SUM=20
Recv: // PWM_AUTO:   00570080 PWM_OFS_AUTO=128 PWM_GRAD_AUTO=87
Recv: // TSTEP:      000fffff TSTEP=1048575

tonn0297, 2019-09-11:
i got it fixed by lowing the 32 microstep to 16 and took run_current to 0.85. dont know why it shuld fuckup the spi but yea.

closed 2019-09-11 by tonn0297

Haschtl, 2019-10-27:
The same missbehaviour appears for me also for the extruders.
And I'm using Marlin-firmware, not klipper. 
I also think this bug wasn't here in some earlier versions of TMCStepper library.

I can tell from my previous problems with the TMC5160, that if microstepping is too high or the maximum mm/s is too high, all motors don't move. Even though I have set the maximum mm/s for the extruder very low, which should work. I will try to set microstepping from 32 to 16 to check if this solves the problem for me, too. 
But I don't think this issue should be closed.