Klipper3d/klipper #1975: Config for Tronxy X1 printer

mfp20, 2019-09-16:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3614968/klippy.log)

I copied the printer-tronxy-x8-2018.cfg but then, [configuring the LCD](https://github.com/KevinOConnor/klipper/issues/1973), I figured out it may be the wrong config (example: wrong heatbed pin). Most of the configs are correct, as I've been able to complete the Configs Check procedure with minor glitches only. But there might be something else to tweak before posting the proper printer-tronxy-x1.cfg file. 

The printer-anet-a8-2017.cfg contains a valid lcd+buttons configuration, but I can't test the heatbed at the moment. 

Please, leave this issue open. I'll attach the correct config at the end, once configured properly, and tested.

mfp20, 2019-09-16:
By not having anything attached to the heatbed sensor pin, the reading is erratic. It may just be the result of a floating pin but as a result, when setting the nozzle temperature to 0 klipper disconnects reporting an unspecified temperature out of range. After setting the heatbed minimum temperature down to -255, setting the nozzle temperature to 0 works properly.

**Change request: keep the nozzles and heatbed temperatures separated so that checks don't make the firmware disconnect.**

Hywelmartin, 2019-09-16:
if you don't have a heated bed, comment it out
```
#[heater_bed]
#heater_pin = PA4
#sensor_type = EPCOS 100K B57560G104F
#sensor_pin = PA6
#control = watermark
#min_temp = -50
#max_temp = 130
```

mfp20, 2019-09-16:
> if you don't have a heated bed, comment it out

I did it before posting the previous comment, and it thrown an error. I couldn't understand wheter it was klipper or the sliced gcode (asking for bed temperature). Anyway...

KevinOConnor, 2019-09-29:
Any further updates on this?

-Kevin

mfp20, 2019-09-30:
Factory PSU burnt, waiting for the new one to arrive ... standby (for the final config file)

closed 2019-11-14 by KevinOConnor

cron410, 2021-06-02:
Here's a good starting point for a nearly stock TronXY X1. I've commented out the heater that I added to a DIY aluminum plate bed.

```
[stepper_x]
step_pin: PD7
dir_pin: PC5
enable_pin: !PD6
microsteps: 16
rotation_distance: 40
endstop_pin: ^!PC2
position_endstop: 0
position_max: 150
homing_speed: 50

[stepper_y]
step_pin: PC6
dir_pin: PC7
enable_pin: !PD6
microsteps: 16
rotation_distance: 40
endstop_pin: ^!PC3
position_endstop: 0
position_max: 150
homing_speed: 50

[stepper_z]
step_pin: PB3
dir_pin: !PB2
enable_pin: !PA5
microsteps: 16
rotation_distance: 8
endstop_pin: ^!PC4
position_endstop: 0.5
position_max: 150

[extruder]
step_pin: PB1
dir_pin: PB0
enable_pin: !PD6
microsteps: 16
rotation_distance: 33.500
#rotation_distance: 33.3325
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PD5
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PA7
control: pid
pid_Kp: 20.16
pid_Ki: 1.30
pid_Kd: 78.02
min_temp: 1
max_temp: 250
#min_extrude_temp: 1
#max_extrude_only_distance: 100

# 12v silicone bed heater #
#[heater_bed]
#heater_pin: PD4
#sensor_type: EPCOS 100K B57560G104F
#sensor_pin: PA6
#control: watermark
#min_temp: 1
#max_temp: 80

[fan]
pin: PB4

[mcu]
serial: /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0

[printer]
kinematics: cartesian
max_velocity: 300
max_accel: 3000
max_z_velocity: 5
max_z_accel: 100

[display]
lcd_type: hd44780
rs_pin: PA3
e_pin: PA2
d4_pin: PD2
d5_pin: PD3
d6_pin: PC0
d7_pin: PC1
up_pin: PA1
analog_range_up_pin: 9000, 13000
down_pin: PA1
analog_range_down_pin: 800, 1300
click_pin: PA1
analog_range_click_pin: 2000, 2500
back_pin: PA1
analog_range_back_pin: 4500, 5000
#kill_pin: PA1
#analog_range_kill_pin: 400, 600
```

cron410, 2021-06-02:
Updated my config in the above comment for Rotation Distance on X and Y, my original value of 40 caused small prints.