Klipper3d/klipper #1977: inline responses from M105/M115 not delivered to host when called from within macro

simonkuehling, 2019-09-16:
Hi Kevin,

I just tried implementing a "M105" loop inside Klipper by using the  [delayed_gcode] module - during that I discovered that responses from commands like M105 and M115 are not pushed through the serial connection to the host, if the command is executed from within a macro.

In the attached klippy.log I ran two tests with M119 and M105 inside a macro to demonstrate the issue. After each one I included a M112 shutdown for a full report in the log.

Besides historical reasons (it has been implemented in other firmwares like that), is there a good case against just changing the response behaviour of these commands to dedicated lines for "ok" and the response string? Like M119 is implemented as well for example.
Otherwise we would at least make sure to carry-over the content of those ack-messages to the host, I guess. 

[klippy-M105-response-in-macro.log](https://github.com/KevinOConnor/klipper/files/3616296/klippy-M105-response-in-macro.log)

KevinOConnor, 2019-09-17:
You could try something like (totally untested):

```patch
--- a/klippy/gcode.py
+++ b/klippy/gcode.py
@@ -650,7 +650,11 @@ class GCodeParser:
     cmd_M105_when_not_ready = True
     def cmd_M105(self, params):
         # Get Extruder Temperature
-        self.ack(self._get_temp(self.reactor.monotonic()))
+        msg = self._get_temp(self.reactor.monotonic())
+        if self.need_ack:
+            self.ack(msg)
+        else:
+            self.respond(msg)
     def cmd_M104(self, params):
         # Set Extruder Temperature
         self._set_temp(params)
```

-Kevin

simonkuehling, 2019-09-18:
That is an excellent idea! Keeps it compatible but still works from macros.
Just tested it - works like a charm!

I think that's worth including into master - what do you think?

KevinOConnor, 2019-09-18:
Okay, thanks.  Commit 8c69e1bc.

-Kevin

closed 2019-09-29 by KevinOConnor