Klipper3d/klipper #1980: Can't change Z_probe

Tomkee, 2019-09-16:
Hi,

just installed klipper and manage to make my BLtouch clone work with some additional commands from instalation guide. But now I have problem with z_offset. No meater what value I will put there my nozzle always end up at the same spot. Am I doing somenthing wrong?

# This file contains common pin mappings for the 2018 Creality
# Ender 3. To use this config, the firmware should be compiled for the
# AVR atmega1284p.

# Note, a number of Melzi boards are shipped with a bootloader that
# requires the following command to flash the board:
#  avrdude -p atmega1284p -c arduino -b 57600 -P /dev/ttyUSB0 -U out/klipper.elf.hex
# If the above command does not work and "make flash" does not work
# then one may need to flash a bootloader to the board - see the
# Klipper docs/Bootloaders.md file for more information.

# See the example.cfg file for a description of available parameters.

[stepper_x]
step_pin: PD7
dir_pin: !PC5
enable_pin: !PD6
step_distance: .0125
endstop_pin: ^PC2
position_endstop: 0
position_max: 235
homing_speed: 50

[stepper_y]
step_pin: PC6
dir_pin: !PC7
enable_pin: !PD6
step_distance: .0125
endstop_pin: ^PC3
position_endstop: 0
position_max: 235
homing_speed: 50

[stepper_z]
step_pin: PB3
dir_pin: PB2
enable_pin: !PA5
step_distance: .0025
endstop_pin:probe:z_virtual_endstop
position_max: 250
position_min:-2

[extruder]
max_extrude_only_distance: 100.0
step_pin: PB1
dir_pin: !PB0
enable_pin: !PD6
step_distance: 0.010526
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PD5
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PA7
control: pid
# tuned for stock hardware with 200 degree Celsius target
pid_Kp: 21.527
pid_Ki: 1.063
pid_Kd: 108.982
min_temp: 0
max_temp: 250

[heater_bed]
heater_pin: PD4
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PA6
control: pid
# tuned for stock hardware with 50 degree Celsius target
pid_Kp: 54.027
pid_Ki: 0.770
pid_Kd: 948.182
min_temp: 0
max_temp: 130

[fan]
pin: PB4

[mcu]
serial: /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AC00FRRN-if00-port0

[printer]
kinematics: cartesian
max_velocity: 300
max_accel: 3000
max_z_velocity: 5
max_z_accel: 100

[display]
lcd_type: st7920
cs_pin: PA3
sclk_pin: PA1
sid_pin: PC1
encoder_pins: ^PD2, ^PD3
click_pin: ^!PC0

[bltouch]
sensor_pin: ^PC4
control_pin: PA4
x_offset: -45
y_offset: -5
z_offset: 2
pin_move_time: 0.4
pin_up_touch_mode_reports_triggered: False

[bed_mesh]
speed: 80
horizontal_move_z: 5
min_point: 90, 20
max_point: 235, 235
probe_count: 3,3
fade_start: 1.0
mesh_pps: 2,2

[gcode_macro G29]
gcode:
    BED_MESH_CALIBRATE

[safe_z_home]
home_xy_position: 155,120
speed: 80.0
z_hop: 10.0
z_hop_speed: 10.0

bgnome, 2019-09-17:
Use probe_calibrate and testz to set your z offset.

closed 2019-09-23 by klipper-gitissuebot

Tomkee, 2019-10-03:
Thank you for anwser. Problem was I didn't save config after calibration SAVE CONFIG command. Hope it will help someone.