Klipper3d/klipper #1982: [feature request] better integration of tmc drivers

tonn0297, 2019-09-17:
hello, i have the tmc5160.
but then printing and the drives report an erroe the printer don't stop? it shuld not be defficult to check the spi every second to see if a driver reports an error. and then just cancel the print.. 
i wuld be a nice feature and somthing even marlin dont hjave i think. in learning python as of right now so im not good enugh to do it yet..
kind regards Tonny

mental405, 2019-09-18:
I feel like this could be combined with 

https://github.com/KevinOConnor/klipper/issues/1934

and whatever the one is about rehoming when a stall is detected which I cant find but I swear it exists.

You are describing some kind of polling timer with callbacks for specific events.

Thermal Warning => Slow down feed rate, turn on fans?
Stall Detected => Save gcode state, rehome
Undervoltage => Halt and catch fire.

tonn0297, 2019-09-18:
yes and no :) the stall wont work.. the stallgaurd is only on when homing. cant be enabled when driving normaly..
and my drivers have undervoltage normaly. if its in err it goes away, dont know about that... but yea. get some handshake from the tmc. so an error is an error..

mental405, 2019-09-18:
Perhaps read the datasheets from Trinamic to understand the capabilities of the drivers and scenarios where various flags are set.

KevinOConnor, 2019-09-18:
This looks like a dup of #1116.

-Kevin

KevinOConnor, 2019-09-29:
I'm going to close this in favour of the existing #1116 - if a developer wishes to implement it then that's great.

-Kevin

closed 2019-09-29 by KevinOConnor