Klipper3d/klipper #1986: M112 emergency stop does not kill G4 dwell (stays unresponsive until dwell timout expires)

simonkuehling, 2019-09-18:
I think I found a bug in emergency stop handling by accidentally running a G4 with super long delay that I tried to abort with M112. While emergency stop did trigger correctly (as shown in the log), the firmware stays unresponsive until the dwell time is expired.

The log file attached is a demonstration of that - and I also tested with shorter dwell time (10s) to see if Klipper is getting responsive again after the timeout passed, which is indeed the case.

[klippy.log](https://github.com/KevinOConnor/klipper/files/3626272/klippy.log)

KevinOConnor, 2019-09-18:
Looks like a dup of #1081.

-Kevin

simonkuehling, 2019-09-19:
Hi @KevinOConnor, this is all tested directly on a serial console (no octoprint) - so I would say it is not a duplicate of #1081 but must be internal to Klipper..

Steps to reproduce:

1. send "G4 6000"
2. send "M112"
3. send "FIRMWARE_RESTART" or any other command like "STATUS" or such

After sending M112 Klipper is correctly shutting down (as evidently in the log file), but stays unresponsive to any new command that is sent. If using a shorter dwell like "G4 S10", you can see that it will get responsive again after the 10 second dwell time is over - so the timer seems to be not cancelled with emergency stop.

KevinOConnor, 2019-09-19:
Your log indicates klipper processed the m112 command within a couple of milliseconds of reading it.  So, it appears the program sending to Klipper delayed the m112.

-Kevin

simonkuehling, 2019-09-19:
That's right, the M112 is beeing executed immediately. It's just that Klipper will not process any command after that - like a "FIRMWARE_RESTART" (for exactly the duration of the G4 command that was under way when M112 fired).

KevinOConnor, 2019-09-21:
I understand.  I'll add it to the list of things to look at, but I likely wont get to it in the near future.

-Kevin

KevinOConnor, 2019-11-07:
This should be fixed now (commit 2d5c7611).

-Kevin

closed 2019-11-14 by KevinOConnor