Klipper3d/klipper #1987: MCU 'mcu' shutdown

Bikeson, 2019-09-18:
Hi, the error started to occur constantly.
MCU 'mcu' shutdown: Rescheduled timer in the past
This generally occurs when the micro-controller has been
requested to step at a rate higher than it is capable of
obtaining.
[klippy.log](https://github.com/KevinOConnor/klipper/files/3627350/klippy.log)

jakep82, 2019-09-18:
The error being reported appears to be correct.  Your microstep and maximum velocity setting are well beyond the capabilities of your MCU.

Bikeson, 2019-09-19:
I printed a few details on these settings. But on this model, such nonsense began. Does not pull what? Is MCU a Ramps microcontroller or motherboard processor?

BlackStump, 2019-09-19:
try dropping your microsteps to 16 instead of 64
adjust your step distance to suit 16 microsteps.
16 microsteps with interpolation is adequate especially with a Ramps controller

Bikeson, 2019-09-19:
Once again, I printed several models precisely on these settings. And where does the steps if Klipper does the calculations. It was because of this that I switched to this firmware in order to unload Ramps and be able to print in 64 steps without interpolation. Or do you want to say that my hardware configuration does not allow me to print with Klipper? Sorry for my language, google translator)

BlackStump, 2019-09-19:
there is still hardware limits on how many steps a ramps controller can generate
ie
https://github.com/KevinOConnor/klipper/blob/master/docs/Features.md#step-benchmarks
if you set on using 64 microsteps then you have to be sure that you print slower to avoid those hardware limits.

Bikeson, 2019-09-19:
Unfortunately, I did not understand from this table how to count on my equipment. The firmware indicates the maximum speed for my hardware. The slicer has a fairly low speed. If I leave 64 steps and turn on interpolation, then should the problem go away?

BlackStump, 2019-09-19:
If I have this right and we use Z only as a example maximum steps is 
154000 *your Z step distance
154000*0.0003125 = 48.125mm/s in real use it would be less then that, if you have any other stepper moving at the same time there is hardware limit exceeded.

Bikeson, 2019-09-19:
Now everything is clear. Thanks for the formula! What about interpolation?

BlackStump, 2019-09-19:
I would use interpolation  as it makes it quieter

jakep82, 2019-09-19:
I will just reiterate what has already been said.  You should use 16x microstepping with interpolation with an AVR (mega2560).  It's not capable of generating pulses fast enough to run 64x microstepping at fast travel speeds, especially on your Y and Z axes.  Enabling interpolation will turn on spreadcycle which will make the printer run much quieter.  It will have no effect on accuracy.

Bikeson, 2019-09-20:
Thank you. So I was wrong, Klipper did not help me increase accuracy. You can close the ticket.

closed 2019-09-29 by KevinOConnor