Klipper3d/klipper #1990: Duet3d in a deadlock

YaronSoffer, 2019-09-19:
Hi,

The log is attached below. 
I am at the early stages of running Klipper on Duet3d/Ender3  ( still haven't done all the work on the config.cfg). I run into a deadlock. There is a need to do "FIRMWARE RESTART" , but i can not do it as the system does not allow me to establish a connection.

Here is the output from the Terminal.
Is there any other way to connect to the Duet3d board and do  "FIRMWARE RESTART ?
=========================================================
Changing monitoring state from "Offline" to "Opening serial port"
Connected to: Serial<id=0x7f4911d090, open=True>(port='/tmp/printer', baudrate=250000, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Starting baud rate detection...
Changing monitoring state from "Opening serial port" to "Detecting baudrate"
Trying baudrate: 250000
Send: N0 M110 N0*125
Recv: // Lost communication with MCU 'mcu'
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Changing monitoring state from "Detecting baudrate" to "Operational"
Send: N0 M110 N0*125
Recv: // Printer is shutdown
Recv: !! Lost communication with MCU 'mcu'
Changing monitoring state from "Operational" to "Error: Lost communication with MCU 'mcu'"
Changing monitoring state from "Error: Lost communication with MCU 'mcu'" to "Offline (Error: Lost communication with MCU 'mcu')"
Connection closed, closing down monitor
Connecting to: /tmp/printer
Changing monitoring state from "Offline" to "Opening serial port"
Connected to: Serial<id=0x7f490771d0, open=True>(port='/tmp/printer', baudrate=250000, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Starting baud rate detection...
Changing monitoring state from "Opening serial port" to "Detecting baudrate"
Trying baudrate: 250000
Send: N0 M110 N0*125
Recv: // Lost communication with MCU 'mcu'
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Changing monitoring state from "Detecting baudrate" to "Operational"
Send: N0 M110 N0*125
Recv: // Printer is shutdown
Recv: !! Lost communication with MCU 'mcu'
Changing monitoring state from "Operational" to "Error: Lost communication with MCU 'mcu'"
Changing monitoring state from "Error: Lost communication with MCU 'mcu'" to "Offline (Error: Lost communication with MCU 'mcu')"
Connection closed, closing down monitor
SendAutoscroll showing 57 lines (Scroll to end) Copy all
Suppress temperature messages
============================================

 cat  /tmp/klippy.log

===== Config file =====
[stepper_x]
step_pin = PD6
dir_pin = PD11
enable_pin = !PC6
step_distance = .0125
endstop_pin = ^PC14
position_endstop = 0
position_max = 250

[tmc2660 stepper_x]
cs_pin = PD14
spi_bus = 1
microsteps = 16
interpolate = True
run_current = 1.000
sense_resistor = 0.051
idle_current_percent = 20

[stepper_y]
step_pin = PD7
dir_pin = !PD12
enable_pin = !PC6
step_distance = .0125
endstop_pin = ^PA2
position_endstop = 0
position_max = 210

[tmc2660 stepper_y]
cs_pin = PC9
spi_bus = 1
microsteps = 16
interpolate = True
run_current = 1.000
sense_resistor = 0.051
idle_current_percent = 20

[stepper_z]
step_pin = PD8
dir_pin = PD13
enable_pin = !PC6
step_distance = .0025
endstop_pin = ^PD29
position_endstop = 0.5
position_max = 200

[tmc2660 stepper_z]
cs_pin = PC10
spi_bus = 1
microsteps = 16
interpolate = True
run_current = 1.000
sense_resistor = 0.051

[extruder0]
step_pin = PD5
dir_pin = PA1
enable_pin = !PC6
step_distance = .002
nozzle_diameter = 0.400
filament_diameter = 1.750
heater_pin = !PA20
sensor_type = EPCOS 100K B57560G104F
sensor_pin = PC15
control = pid
pid_kp = 22.2
pid_ki = 1.08
pid_kd = 114
min_temp = 0
max_temp = 250

[tmc2660 extruder0]
cs_pin = PC17
spi_bus = 1
microsteps = 16
interpolate = True
run_current = 1.000
sense_resistor = 0.051

[heater_bed]
heater_pin = !PA19
sensor_type = EPCOS 100K B57560G104F
sensor_pin = PC13
control = watermark
min_temp = 0
max_temp = 120

[fan]
pin = PC23

[heater_fan nozzle_cooling_fan]
pin = PC26
heater = extruder0
heater_temp = 45
fan_speed = 1.0

[mcu]
serial = /dev/serial/by-id/usb-Klipper_Klipper_firmware_12345-if00
restart_method = command

[printer]
kinematics = cartesian
max_velocity = 300
max_accel = 3000
max_z_velocity = 5
max_z_accel = 100

[static_digital_output onboard_led]
pins = !PC2
=======================
Loaded MCU 'mcu' 77 commands (v0.7.0-349-gd766587-20190912_151942-ender3 / gcc: (15:4.9.3+svn231177-1) 4.9.3 20150529 (prerelease) binutils: (2.26-4ubuntu1+8) 2.26)
MCU 'mcu' config: ADC_MAX=4095 PWM_MAX=255 CLOCK_FREQ=120000000 SOFT_PWM_MAX=256 STEP_DELAY=2 MCU=sam4e8e STATS_SUMSQ_BASE=256
Configured MCU 'mcu' (1024 moves)
Args: ['/home/pi/klipper/klippy/klippy.py', '/home/pi/printer.cfg', '-l', '/tmp/klippy.log']
Git version: 'v0.7.0-349-gd766587'
CPU: 4 core ?
Python: '2.7.12 (default, Aug 22 2019, 16:36:40) \n[GCC 5.4.0 20160609]'
=============== Log rollover at Thu Sep 19 19:51:53 2019 ===============
Lost communication with MCU 'mcu'
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown

Lost communication with MCU 'mcu'
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown

BlackStump, 2019-09-19:
why do you not use the duet2.cfg as is to get connected first? 
ie
https://github.com/KevinOConnor/klipper/blob/master/config/generic-duet2.cfg

FHeilmann, 2019-09-20:
type `FIRMWARE_RESTART` into your octoprint terminal. I also recommend getting the OctoKlipper addon which will give you some nice buttons to do that stuff.

YaronSoffer, 2019-09-22:
Hello all, 
I can not type anything as there is no way to type ( the machine is not in "connect"  mode and therefore the terminal is in a greied out mode). 
The problem is not in the CFG, but as I see it in the behavior of the octoklipper plugin octoprint.
Even though the firmware software needs a "FIRMWARE_RESTART" , the plugin should move to a "connected mode" allowing me to send "FIRMWARE_RESTART" through the terminal, and not disconnect itself.

BlackStump, 2019-09-22:
@YaronSoffer 
I am confused about what your problem is and how it perceived as a klipper firmware problem,
The instructions for installation are pretty clear.
If your problem is Octoprint or Octoklipper  then you would be better served reporting your issues to Octoprint and or Octoklipper?
If it is a install config problem then these instructions will explain how to ssh into your pi where your can issue a firmware_restart and retrieve your log file.

>After installing OctoPi and upgrading OctoPrint, it will be necessary to ssh into the target machine to >run a handful of system commands. If using a Linux or MacOS desktop, then the "ssh" software >should already be installed on the desktop. There are free ssh clients available for other desktops 
>(eg, PuTTY). Use the ssh utility to connect to the Raspberry Pi

YaronSoffer, 2019-09-23:
**@BlackStump**
I can not connect to the board due to the fact it needs a "FIRMWARE_RESTART" ( The interface won't let me, it detect that the firmware needs a "FIRMWARE_RESTART" - and close the connection automatically, before I can type down anything). I can not apply a "FIRMWARE_RESTART"  because I can not connect to the board. This is why it says deadlock in the title.

Here are some snaps from the session:
Changing monitoring state from "Offline" to "Opening serial port" ....
Changing monitoring state from "Opening serial port" to "Detecting baudrate" ...Trying baudrate: 250000 .....
Recv: // **"FIRMWARE_RESTART" command to reset the firmware, reload the**
Recv: // config, and restart the host software.
Changing monitoring state from "Detecting baudrate" to "Operational"
Send: N0 M110 N0125
Recv: // Printer is shutdown
Recv: **!! Lost communication with MCU 'mcu'**
Changing monitoring state from "Operational" to "Error: Lost communication with MCU 'mcu'"

Is there any other way to send "FIRMWARE_RESTART" to the firmware  ?

BlackStump, 2019-09-23:
https://github.com/KevinOConnor/klipper/blob/master/docs/Installation.md#configuring-octoprint-to-use-klipper
read carefully this section, have you done this in octoprint?
>In the Settings tab, navigate to the "Behavior" sub-tab and select the "Cancel any ongoing prints but >stay connected to the printer" option. Click "Save".
the other option is to shh into your pi and issue "sudo service klipper stop" then "sudo service klipper start"
you also need to investigate why you losing communications ie printer is shutdown

YaronSoffer, 2019-09-23:
This one helped 👍 

Using @BlackStump tip ( thanks).
Do the following from:
https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#i-keep-getting-random-lost-communication-with-mcu-errors .

I am connected to the printer, but I had to cancel the connection through the octoklipper plugin. I am using the default octoprint connection ( uncheck in the octoklipper setting plugin the "Replace Connection Panel". Also, my connection parameters are as follow:
Serial port - "/tmp/printer"
Baudrate  - "AUTO".

My configuration 
DUET3D board (cloned)  + Orange Pi  + ENDER3. 
Hope it will help someone else ...

closed 2019-09-23 by YaronSoffer