Klipper3d/klipper #1991: Voltage problem with TSL1401CL to SKR 1.3

terrorhai, 2019-09-19:
Hey guys,

was planning to reattach my TSL1401CL sensor for filament width measuring. This sensor gives back the filament width in 1V per mm, so 1.75mm filament is 1.75V. It's driven by 5V, so if you don't got any filament in it, it jumps between 1V and 5V.

The Problem is, that the SKR (or all LPC1768) allow 5V on the input pins (can use endstop pins), but NOT in analog mode. Then it's max input voltage is 3.3V...you will probably get what I want to say: If the filament is in and measured, everything is fine. BUT if you put it out, it will start jumping to 5V and will kill the analog input and so the CPU (probably).

1st solution: Just put in the stock board as a second MCU and use it just for the filament sensor, which is like driving to the supermarket with a 40 ton truck or

2nd solution: cut the voltage to a max of 3.3V...and that's my problem, I'm not an electronics pro (can solder well and all that stuff, wire stuff up, know what + and - is, but that's it).
So is anyone out there who probably knows how to fix this problem?

I heared of a diode called "zener diode", does that fix the problem?

Log is just a cookie for the bot :)

terrorhai, 2019-09-19:
[klippy.log](https://github.com/KevinOConnor/klipper/files/3633107/klippy.log)

jakep82, 2019-09-19:
You'll have better luck asking this question in a more appropriate place like electronics section of the RepRap forums.  This has nothing to do with Klipper, or even software in general.

terrorhai, 2019-09-20:
I found a solution with Jason who developed this sensor in comination with a Attiny 85. Probably it will help someone to not destroy his SKR board ;)
[image]

FHeilmann, 2019-09-20:
looking at the [datasheet of the TSL1401CL](https://www.mouser.de/datasheet/2/588/TSL1401CL-1214741.pdf), it accepts supply voltage all the way down to 3V, so you could theoretically run it at 3.3V and it won't screw up your SKR input pin.
Whats weird to me  is that the datasheet (page 3) also states that that Vout can only reach 1.6-2.4V. So unless your sensor has additional circuitry I don't see how 5V at the output would be possible.

mental405, 2019-09-20:
I am like, the only other person in the world with this sensor. I have not hooked it up to my SKR yet. Good catch on the pin. I just confirmed in the datasheet, when a pin is switched to analog, it is no longer 5V tolerant.

I am seeing that the ATTiny85 accepts an input voltage from 2.7-5.5V>

As I have a couple of these modules lying about on my desk, I am going to hook one up to my scope and power it with 3.3V off my bench supply. If the analog signal coming back does not peak about the input voltage and the output voltage is correct, you should be fine just running it at 3.3V instead of 5V. Additionally I have a 1.75mm pin gauge that I can use to calibrate it as well to ensure consistent readings.

nophead, 2019-09-20:
The sensor seems to be 1D camera with an analogue output. How does that give a voltage proportional to the filament width? Does one simply average it so the number of pixels covered versus those not gives PWM signal with a mark space ratio proportional to filament width?

I am still surprised it 1.75V equals 1.75mm. Is there some further signal processing?

mental405, 2019-09-20:
There's an ATtiny85 or similar microcontroller running code  that processes the optical array from the sensor and determines how wide the filament is based off of how many sensors are occluded. 

It's then takss that value and outputs it as a voltage in the form of a pwm signal from one of the pins.

Klipper takes that voltage and and stores it in a buffer based off of how many millimeters of filament are between the extruder and the sensor.

When an extrusion move is being calculated it locates the appropriate value in the buffer and applies it to the requested extrusion length.

nophead, 2019-09-20:
Nice. Shame the output isn't digital.

mental405, 2019-09-20:
It is a shame. I had to put a capacitor on mine to filter the signal because it fluctuated so much on my old board.

I suspect I am going to have to pilfer the smoothing code from whatever the thermistor is doing to get a better average.

I started writing in support for it to double as a filament runout sensor but got sidetracked and never got back around to that either.

terrorhai, 2019-09-22:
> 
> 
> It is a shame. I had to put a capacitor on mine to filter the signal because it fluctuated so much on my old board.
> 
> I suspect I am going to have to pilfer the smoothing code from whatever the thermistor is doing to get a better average.
> 
> I started writing in support for it to double as a filament runout sensor but got sidetracked and never got back around to that either.
> 
> Another issue with it is that it only accounts for one direction. If the cross section is elliptical, it wouldn't measure that.

There is a solution with 3 LEDs in 45° angles to eleminate that problem. But I still go with a 1 LED solution. Will probably change that but actually I see no need to do that.

KevinOConnor, 2019-10-08:
I'm going to close this issue as it appears the conversation has concluded.

-Kevin

closed 2019-10-08 by KevinOConnor