Klipper3d/klipper #1992: Help me understand why bed mesh probing is not respecting bl touch offset?

CalielOfSeptem, 2019-09-20:
I am new to klippy, and semi-new to 3d printers in general.  I have most everything working on my modified Ender 5.  The big change is the SKR pro v1.1 board, most everything else is essentially stock but-for the BL touch.

In any event, probing works great until I try and run BED_MESH_CALIBRATE.  Starts at 0,0 but then eventually moves to an X position that is off the print bed (see fuzzy attached pic).  My offsets & limits are correct (at least I think so), see the attached log file.  These same offsets worked in my prior configuration where I used a Duet Wifi (without klipper).  I re-measured them based on Kevin's guide and they are the same offsets....

What is also weird is if I move the nozzle to X0,Y0, and then do a 'probe', the result comes back saying my probe is at 0,0.  Maybe that is by design... but I would expect the offset to change that value to 42,16 based on my bltouch section.

Please help me understand what it is I am doing wrong. 

[bad_klippy_probe.txt](https://github.com/KevinOConnor/klipper/files/3637306/bad_klippy_probe.txt)

[image]

jakep82, 2019-09-20:
All coordinates are in reference to the nozzle position, and probe offsets are not used when establishing min and max probe points, only when calculating the mesh itself.  See the bed_mesh documentation in example-extras.cfg for more information (linked below).  Also note that a PR is in process to change this to avoid confusion.  This has been asked and answered many times.

https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L54
https://github.com/KevinOConnor/klipper/pull/1949

CalielOfSeptem, 2019-09-20:
@jakep82 Thanks for the response.  Seems like simply having the BL touch/probe documentation mention this fact would make a lot of sense... unless it already does.  I will adjust the parameters knowing the offsets are not being applied.

jakep82, 2019-09-20:
It is documented in example-extras.cfg which contains all of the configuration parameters, and an explanation of each of those parameters.  Even so, it's confusing to many people which is why it's being changed.

jalanjarosz, 2019-09-20:
Mesh Leveling does not take into consideration the BLTouch offset, like Marlin does.

Here is my config for BLTouch AND Mesh.

```
[bltouch] # clone
sensor_pin: ^z:ar10
control_pin: z:ar11
pin_move_time: 0.675
pin_up_reports_not_triggered: False
pin_up_touch_mode_reports_triggered: False
x_offset: 28.5  # Distance to the right of the nozzle
y_offset: 0 # Distance in front of the nozzle
z_offset: 1.80
speed: 5.0
samples: 3
sample_retract_dist:5
samples_result:median
#test_sensor_pin: False
samples_tolerance: 0.15
samples_tolerance_retries :2
```

```
[bed_mesh]
speed: 50
horizontal_move_z: 15
min_point: -18.5,10  # Nozzle zero to get probe at a 10,10 position.
max_point: 261.5,290 # Nozzle position to get probe to a 290,290 position of a 300x300 bed
probe_count: 5,5
fade_start: 1.0
fade_end: 10.0
split_delta_z: .025
move_check_distance: 5.0
mesh_pps: 2,2
algorithm: lagrange
bicubic_tension: .1
```

Hope this helps.

CalielOfSeptem, 2019-09-21:
> Mesh Leveling does not take into consideration the BLTouch offset, like Marlin does.
> 
> Here is my config for BLTouch AND Mesh.
> 
> ```
> [bltouch] # clone
> sensor_pin: ^z:ar10
> control_pin: z:ar11
> pin_move_time: 0.675
> pin_up_reports_not_triggered: False
> pin_up_touch_mode_reports_triggered: False
> x_offset: 28.5  # Distance to the right of the nozzle
> y_offset: 0 # Distance in front of the nozzle
> z_offset: 1.80
> speed: 5.0
> samples: 3
> sample_retract_dist:5
> samples_result:median
> #test_sensor_pin: False
> samples_tolerance: 0.15
> samples_tolerance_retries :2
> ```
> 
> ```
> [bed_mesh]
> speed: 50
> horizontal_move_z: 15
> min_point: -18.5,10  # Nozzle position to get probe at a 10,10 position.
> max_point: 261.5,290 # Nozzle position to get probe to a 290,290 position of a 300x300 bed
> probe_count: 5,5
> fade_start: 1.0
> fade_end: 10.0
> split_delta_z: .025
> move_check_distance: 5.0
> mesh_pps: 2,2
> algorithm: lagrange
> bicubic_tension: .1
> ```
> 
> One would think that the Mesh would use the probe offset, but it does not.
> 
> Hope this helps.

Thanks! That is helpful

KevinOConnor, 2019-09-21:
> Mesh Leveling does not take into consideration the BLTouch offset,

Mesh bed levelling does take into account the probe offset.  The probing region is currently specified in nozzle coordinates, however.  It is understood that this can be confusing so a change to use "bed coordinates" instead of "nozzle coordinates" is in discussion - PR #1949.

-Kevin

KevinOConnor, 2019-09-29:
I'm going to close this issue in favour of the existing #1949.

-Kevin

closed 2019-09-29 by KevinOConnor