Klipper3d/klipper #1993: float division by zero

tonn0297, 2019-09-20:
Hello @KevinOConnor 
i have a problem with my printer. this error is the first time i've encountered. i hope you can see what caused it. 

[klippy.zip](https://github.com/KevinOConnor/klipper/files/3637846/klippy.zip)

i don't know if i can reproduce it. 
i'm running:
tmc5160 
skr v1.3
kipper work-debug-20190908

tonn0297, 2019-09-22:
and im running klipper work-debug bc the fix for my bl touch is not in the main.. or at least it don't work.
kind regards

FHeilmann, 2019-09-23:
The float division by zero error was caused in the `web_dwc2.py` extra, which is not part of mainline klipper. Show this error to the guys over at the dwc2 for klipper repo so they can fix it for you :) Make sure to close this issue as well.

KevinOConnor, 2019-09-29:
Okay, I'm going to close this issue as this issue tracker is only for the pristine code found in this github repo.

-Kevin

closed 2019-09-29 by KevinOConnor

Stephan3, 2019-09-30:
@tonn0297 checkout latest master of dwc-for-klipper.