Klipper3d/klipper #1995: 4X uStepperS, 328PB, TMC5130 and trying to load Klipper

dingbatca, 2019-09-21:
Feature request, Processor Model, Stepper driver, and/or example config request, I am not sure. 

I am not sure where best to ask for help.  Brand new to Klipper, so please be gentile.  

Supporting links:
https://ustepper.com/store/ustepper-boards/27-48-ustepper-s.html
https://github.com/uStepper/uStepperS/

Delta machine, 3 steppers + one for the extruder.  Would like 1X uStepperS per motor.  Yes, I am aware that flashing Klipper will loose the encoder/closed loop.  Just treat this problem like a 328PB hard wired to a TMC5130.

Trying to load the Klipper firmware onto a uStepperS. The MCU on the uStepperS is a 328PB, and I only see the 328P in the menu.  Not sure if this is an issue, yet.  I know the stock 328P does not have a PE3 pin, which is required for MOSI.

Here is my reverse engineering of uStepperS pin map. Please double check.
cs_pin: PE2
spi_software_sclk_pin: PC1
spi_software_mosi_pin: PE3
spi_software_miso_pin: PC0
step_pin: PB2
dir_pin: PB1
enable_pin: PD4

I think my config should look like this, for stepper_a/b/c/extruder:
```
[stepper_x]
step_pin: PB2
dir_pin: PB1
enable_pin: !PD4, tmc5160_stepper_x:virtual_enable
step_distance: .0125
endstop_pin: ???? PD2 (Using ping D3 on the uStepperS for a simple limit switch)
position_endstop: 0
position_max: 250

[tmc5160 stepper_x]
cs_pin: PE2
spi_bus: spi????????
microsteps: 16
run_current: 1.000
sense_resistor: ?????
```

Last part, and I am not even sure how to ask it.  How do I call out each stepper_x as it's own USB attached MCU?

Thanks for helping
Adam

cosminr86, 2019-09-21:
You must flash every ustepper with klipper.

Take a look at multimcu example config file. You must define an mcu for every stepper and connect all steppers to the pi

KevinOConnor, 2019-09-21:
It appears that the 328P code should run on the 328PB.  You wont have access to the PE3 pin with the current code.  However, it is not difficult to add support for that.  If you can get the current code installed (and provide a klipper log file) I can work with you to get the additional pins supported.

As mentioned above, use multiple mcu sections for each uStepper.  See the https://github.com/KevinOConnor/klipper/blob/master/config/example-multi-mcu.cfg file for an example of this.

-Kevin

dingbatca, 2019-09-21:
I figured I would need the correct firmware per MCU. ;-)

Had to force the flash:
```
pi@octopi:~/klipper $ make flash FLASH_DEVICE=/dev/serial/by-id/usb-ON_Development_IVS_uStepper_S_Lit_de2a585c87a3e8119efb93603b0549ec-if00-port0
  Flashing out/klipper.elf.hex to /dev/serial/by-id/usb-ON_Development_IVS_uStepper_S_Lit_de2a585c87a3e8119efb93603b0549ec-if00-port0 via avrdude

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9516 (probably m328pb)
avrdude: Expected signature for ATmega328P is 1E 95 0F
         Double check chip, or use -F to override this check.

avrdude done.  Thank you.

src/avr/Makefile:32: recipe for target 'flash' failed
make: *** [flash] Error 1
```

The above fail was expected, so I forced it:
```
pi@octopi:~/klipper $ avrdude -F -p m328p -c arduino -P /dev/serial/by-id/usb-ON_Development_IVS_uStepper_S_Lit_de2a585c87a3e8119efb93603b0549ec-if00-port0 flash:w:out/klipper.elf.hex

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9516 (probably m328pb)
avrdude: Expected signature for ATmega328P is 1E 95 0F

avrdude: safemode: Fuses OK (E:00, H:00, L:00)

avrdude done.  Thank you.

pi@octopi:~/klipper $ echo $?
0
```

Looks like I am unable to connec to the re-flashed 328PB (with the 328P code).  Only part I have configured is mcu xboard, and stepper x.
```Starting Klippy...
Args: ['/home/pi/klipper/klippy/klippy.py', '/home/pi/printer.cfg', '-l', '/tmp/klippy.log']
Git version: 'v0.7.0-864-g626cb714'
CPU: 4 core ARMv7 Processor rev 4 (v7l)
Python: '2.7.13 (default, Sep 26 2018, 18:42:22) \n[GCC 6.3.0 20170516]'
Start printer at Sat Sep 21 12:41:03 2019 (1569094863.9 336361.0)
===== Config file =====
[mcu]
serial = /dev/serial/by-id/usb-ON_Development_IVS_uStepper_S_Lit_de2a585c87a3e8119efb93603b0549ec-if00-port0
pin_map = arduino

[mcu xboard]
serial = /dev/serial/by-id/usb-ON_Development_IVS_uStepper_S_Lit_de2a585c87a3e8119efb93603b0549ec-if00-port0
pin_map = arduino

[stepper_x]
step_pin = xboard:PB2
dir_pin = xboard:PB1
enable_pin = !xboard:PD4, tmc5160_stepper_x:virtual_enable
step_distance = .0125
endstop_pin = ^xboard:PC14
position_endstop = 0
position_max = 250

[tmc5160 stepper_x]
cs_pin = xboard:PE2
spi_bus = spi
microsteps = 16
run_current = 1.000
[... 78 lines cut ...]
    raise error("Timeout on wait for '%s' response" % (self.name,))
error: Timeout on wait for 'identify_response' response
Timeout on serial connect
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/serialhdl.py", line 92, in connect
    identify_data = self._get_identify_data(connect_time + 5.)
  File "/home/pi/klipper/klippy/serialhdl.py", line 57, in _get_identify_data
    params = self.send_with_response(msg, 'identify_response')
  File "/home/pi/klipper/klippy/serialhdl.py", line 156, in send_with_response
    return src.get_response([cmd], self.default_cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 225, in get_response
    raise error("Timeout on wait for '%s' response" % (self.name,))
error: Timeout on wait for 'identify_response' response
Timeout on serial connect
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/serialhdl.py", line 92, in connect
    identify_data = self._get_identify_data(connect_time + 5.)
  File "/home/pi/klipper/klippy/serialhdl.py", line 57, in _get_identify_data
    params = self.send_with_response(msg, 'identify_response')
  File "/home/pi/klipper/klippy/serialhdl.py", line 156, in send_with_response
    return src.get_response([cmd], self.default_cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 225, in get_response
    raise error("Timeout on wait for '%s' response" % (self.name,))
error: Timeout on wait for 'identify_response' response
```

cosminr86, 2019-09-21:
Your config is wrong.
You have defined 2 Mcu's but they are on the same USB port ?!

dingbatca, 2019-09-22:
My config is very wrong.  It is also in a great amount of flux! I am new to klipper, not config files. ;-)

Finally got everything to flash correctly.  Got the error I was expecting "Unknown value 'PE2'":
```
=======================
Starting serial connect
got {u'count': 115, '#receive_time': 359934.51534899767, u'sum': 209194, u'sumsq': 2373260, '#name': u'stats', '#sent_time': 359934.51011154935}
Loaded MCU 'mcu' 80 commands (v0.7.0-864-g626cb714-20190921_191024-octopi / gcc: (GCC) 5.4.0 binutils: (GNU Binutils) 2.26.20160125)
MCU 'mcu' config: RECEIVE_WINDOW=192 BUS_PINS_spi=PB4,PB3,PB5 SERIAL_BAUD=115200 STEP_DELAY=-1 ADC_MAX=1023 PWM_MAX=255 RESERVE_PINS_serial=PD0,PD1 BUS_PINS_twi=PC5,PC4 MCU=atmega328p STATS_SUMSQ_BASE=256 CLOCK_FREQ=16000000
Sending MCU 'mcu' printer configuration...
Protocol error
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 138, in _connect
    cb()
  File "/home/pi/klipper/klippy/mcu.py", line 672, in _connect
    self._check_config()
  File "/home/pi/klipper/klippy/mcu.py", line 613, in _check_config
    self._send_config(None)
  File "/home/pi/klipper/klippy/mcu.py", line 587, in _send_config
    self._serial.send(c)
  File "/home/pi/klipper/klippy/serialhdl.py", line 151, in send
    cmd = self.msgparser.create_command(msg)
  File "/home/pi/klipper/klippy/msgproto.py", line 330, in create_command
    cmd = mp.encode_by_name(**argparts)
  File "/home/pi/klipper/klippy/msgproto.py", line 156, in encode_by_name
    t.encode(out, params[name])
  File "/home/pi/klipper/klippy/msgproto.py", line 109, in encode
    v, self.enum_name))
error: Unknown value 'PE2' in enumeration 'pin'
```

KevinOConnor, 2019-09-24:
Unfortunately, it looks like the standard avr-libc toolchain has not been updated for the 328pb.  That makes supporting this chip a pain.  That said, we can define the PEx pins - if you grab the code at:

cd ~/klipper ; git fetch ; git checkout origin/work-atmega328pb-20190924 ; sudo service klipper stop ; make ; ... flash ... ; sudo service klipper start

Does that work?

-Kevin

dingbatca, 2019-09-24:
Kinda.  Code compiles and I got Klipper up and running + 5 MCU's flashed.  ABC motors of my delta are attached.  G28 failed to home.  To be exact the motors never enable.  I can spin the drive shafts by hand.

I would guess:

1. My SPI settings are wrong?
2. I failed to properly reverse engineer the schematics of the uStepperS board?
3. The support for the new pins is not working?
4. The TMC5130 are different then the TMC5160 when setting up?

I will double check #2 tonight.  I would really appreciate it if some one could check my below config ,#1, and make sure I got the SPI settings correct. 

Thanks!

```
pi@octopi:~ $ cat /tmp/klippy.log  | head -n 200
Starting Klippy...
Args: ['/home/pi/klipper/klippy/klippy.py', '/home/pi/printer.cfg', '-l', '/tmp/klippy.log']
Git version: 'v0.7.0-868-g84c16a6e'
CPU: 4 core ARMv7 Processor rev 4 (v7l)
Python: '2.7.13 (default, Sep 26 2018, 18:42:22) \n[GCC 6.3.0 20170516]'
Start printer at Tue Sep 24 12:35:24 2019 (1569353724.2 595222.4)
===== Config file =====
[mcu]
serial = /dev/ttyACM0
pin_map = arduino
baud = 115200

[mcu aboard]
serial = /dev/serial/by-id/usb-ON_Development_IVS_uStepper_S_Lit_3a46ff5489a3e81192f992603b0549ec-if00-port0
pin_map = arduino
baud = 115200

[mcu bboard]
serial = /dev/serial/by-id/usb-ON_Development_IVS_uStepper_S_Lit_4048b1dfa3a3e811970794603b0549ec-if00-port0
pin_map = arduino
baud = 115200

[mcu cboard]
serial = /dev/serial/by-id/usb-ON_Development_IVS_uStepper_S_Lit_b69e64bdef13e811860a9d3a3b0549ec-if00-port0
pin_map = arduino
baud = 115200

[mcu eboard]
serial = /dev/serial/by-id/usb-ON_Development_IVS_uStepper_S_Lit_de2a585c87a3e8119efb93603b0549ec-if00-port0
pin_map = arduino
baud = 115200

[stepper_a]
step_pin = aboard:PB2
dir_pin = !aboard:PB1
enable_pin = !aboard:PD4, tmc5160_stepper_a:virtual_enable
step_distance = .01
endstop_pin = !aboard:PD2
homing_speed = 50
position_endstop = 512
arm_length = 304.11

[stepper_b]
step_pin = bboard:PB2
dir_pin = !bboard:PB1
enable_pin = !bboard:PD4, tmc5160_stepper_b:virtual_enable
step_distance = .01
endstop_pin = !bboard:PD2

[stepper_c]
step_pin = cboard:PB2
dir_pin = !cboard:PB1
enable_pin = !cboard:PD4, tmc5160_stepper_c:virtual_enable
step_distance = .01
endstop_pin = !cboard:PD2
[... 73 lines cut ...]
Sending MCU 'bboard' printer configuration...
Configured MCU 'bboard' (116 moves)
Starting serial connect
got {'#receive_time': 595226.5356813236, u'crc': 3437396119L, u'is_shutdown': 0, '#name': u'config', u'is_config': 1, '#sent_time': 595226.5308587892, u'move_count': 116}
Loaded MCU 'cboard' 80 commands (v0.7.0-868-g84c16a6e-20190924_075344-octopi / gcc: (GCC) 5.4.0 binutils: (GNU Binutils) 2.26.20160125)
MCU 'cboard' config: RECEIVE_WINDOW=192 BUS_PINS_spi=PB4,PB3,PB5 SERIAL_BAUD=115200 STEP_DELAY=-1 ADC_MAX=1023 PWM_MAX=255 RESERVE_PINS_serial=PD0,PD1 BUS_PINS_twi=PC5,PC4 MCU=atmega328p STATS_SUMSQ_BASE=256 CLOCK_FREQ=16000000
Sending MCU 'cboard' printer configuration...
Configured MCU 'cboard' (116 moves)
Starting serial connect
got {'#receive_time': 595227.8926034586, u'crc': 3437396119L, u'is_shutdown': 0, '#name': u'config', u'is_config': 1, '#sent_time': 595227.8882723303, u'move_count': 116}
got {'#receive_time': 595228.1297564277, '#name': u'starting', '#sent_time': 0.0}
got {'#receive_time': 595228.5544874168, '#name': u'starting', '#sent_time': 0.0}
got {'#receive_time': 595228.5561255937, '#name': u'starting', '#sent_time': 0.0}
got {'#receive_time': 595228.7112920007, '#name': u'starting', '#sent_time': 0.0}
Loaded MCU 'eboard' 80 commands (v0.7.0-868-g84c16a6e-20190924_075344-octopi / gcc: (GCC) 5.4.0 binutils: (GNU Binutils) 2.26.20160125)
MCU 'eboard' config: RECEIVE_WINDOW=192 BUS_PINS_spi=PB4,PB3,PB5 SERIAL_BAUD=115200 STEP_DELAY=-1 ADC_MAX=1023 PWM_MAX=255 RESERVE_PINS_serial=PD0,PD1 BUS_PINS_twi=PC5,PC4 MCU=atmega328p STATS_SUMSQ_BASE=256 CLOCK_FREQ=16000000
Sending MCU 'eboard' printer configuration...
Configured MCU 'eboard' (112 moves)
got {'#receive_time': 595229.2140267396, u'crc': 3676893733L, u'is_shutdown': 0, '#name': u'config', u'is_config': 1, '#sent_time': 595229.2108551596, u'move_count': 112}
Starting heater checks for extruder
Stats 595229.2: gcodein=0 mcu: mcu_awake=0.002 mcu_task_avg=0.000035 mcu_task_stddev=0.000045 bytes_write=614 bytes_read=3324 bytes_retransmit=9 bytes_invalid=0 send_seq=81 receive_seq=81 retransmit_seq=2 srtt=0.004 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=15982451 aboard: mcu_awake=0.016 mcu_task_avg=0.000116 mcu_task_stddev=0.000112 bytes_write=793 bytes_read=3391 bytes_retransmit=9 bytes_invalid=0 send_seq=105 receive_seq=105 retransmit_seq=2 srtt=0.006 rttvar=0.002 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=15959202 adj=15968807 bboard: mcu_awake=0.016 mcu_task_avg=0.000116 mcu_task_stddev=0.000112 bytes_write=781 bytes_read=3336 bytes_retransmit=23 bytes_invalid=0 send_seq=103 receive_seq=97 retransmit_seq=97 srtt=0.006 rttvar=0.002 rto=0.100 ready_bytes=0 stalled_bytes=0 freq=15953252 adj=15916552 cboard: mcu_awake=0.016 mcu_task_avg=0.000119 mcu_task_stddev=0.000114 bytes_write=775 bytes_read=3334 bytes_retransmit=0 bytes_invalid=0 send_seq=103 receive_seq=97 retransmit_seq=0 srtt=0.006 rttvar=0.002 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=15949723 adj=15964011 eboard: mcu_awake=0.000 mcu_task_avg=0.000000 mcu_task_stddev=0.000000 bytes_write=786 bytes_read=3286 bytes_retransmit=0 bytes_invalid=0 send_seq=101 receive_seq=95 retransmit_seq=0 srtt=0.007 rttvar=0.003 rto=0.025 ready_byte [... cut ...]
MCU 'cboard' shutdown: Invalid oid type
clocksync state: mcu_freq=16000000 last_clock=4297287247 clock_est=(595227.817 77638629 15949723.089) min_half_rtt=0.001729 min_rtt_time=595227.869 time_avg=595227.815(0.000) clock_avg=77638629.134(4891.293) pred_variance=188889552.616 clock_adj=(10242.927 15968616.500)
Dumping serial stats: bytes_write=775 bytes_read=3398 bytes_retransmit=0 bytes_invalid=0 send_seq=103 receive_seq=102 retransmit_seq=0 srtt=0.006 rttvar=0.002 rto=0.025 ready_bytes=0 stalled_bytes=0
```

dingbatca, 2019-09-24:
Whoops!  Needed the SPI settings for all motors in the config.  But still no life.

```
[tmc5160 stepper_a]
cs_pin: aboard:PE2
#spi_bus: spi
microsteps: 16
run_current: 1.000
sense_resistor: 0.051
spi_software_sclk_pin: aboard:PC1
spi_software_mosi_pin: aboard:PE3
spi_software_miso_pin: aboard:PC0
```

KevinOConnor, 2019-09-25:
Do you get valid results from the DUMP_TMC command?  (See docs/G-Codes.md for the details.)

-Kevin

dingbatca, 2019-09-26:
```
Send: status
Recv: // Klipper state: Ready
Recv: ok
Send: DUMP_TMC
Recv: !! Error on 'DUMP_TMC': missing STEPPER
```

Again, please double check my pin out and SPI settings.  I am NOT sure of these:
https://github.com/uStepper/uStepperS/blob/master/Hardware/Schematics.PDF

I know NOTHING about SPI.  Here is what I think I know for pin mapping:
```
SCK1=PC1
MISO1=PC0
MOSI1=PE3
CLK0=PB0
CS1_1=PE2
CLK0=PB0
DIR=PB1
STP=PB2
ENA=PD4
```

Here is my configuration based on the above info:
```
[stepper_b]
step_pin: bboard:PB2
dir_pin: !bboard:PB1
enable_pin: !bboard:PD4, tmc5160_stepper_b:virtual_enable
step_distance: .01
endstop_pin: !bboard:PD2

[tmc5160 stepper_e]
cs_pin: eboard:PE2
#spi_bus: spi
microsteps: 16
run_current: 1.000
sense_resistor: 0.051
spi_software_sclk_pin: eboard:PC1
spi_software_mosi_pin: eboard:PE3
spi_software_miso_pin: eboard:PC0
```

dingbatca, 2019-09-28:
As much as I would love to see the uStepperS's work.  Lets close this thread out.  I ordered an SKR Pro with 4X TMC5160's.  Should be here in a few weeks.

I started with Smoothieware, then Marlin, and now Klipper.  LOVING the Klipper firmware.  KevinOConnor, thanks for your amazing work in creating such a kick butt firmware.

KevinOConnor, 2019-09-29:
Okay - feel free to close this issue.  FYI, the usage of the DUMP_TMC command is documented in the docs/G-Codes.md file.  Alas, I don't have any familiarity with the uStepper, so can't provide any meaningful advice on pin mappings.

-Kevin

closed 2019-09-30 by dingbatca