Klipper3d/klipper #1996: To flash or not to flash?

karabas2011, 2019-09-21:
Mks Sbase and PI3+  via USB

1. How can I flash sbase without physical accessing microsd ? On some printers it's rather hard. 

2 Does hardware serial coonection work better usb and more reliable?
I tried to connect via hardware serial but failed. Is it possible? I used to use this in smoothieware setup. I blocked bluetooth  in pi3+ and connected via /dev/ttyAMA0

3 Can I use any other serial terminals like minicom to connect to /tmp/printer insteed of octoprint ( for debugging)

karabas2011, 2019-09-21:
Sorry, I have solved #1 , never thought sbase is exactly the same usb ID as smoothieboard.

FHeilmann, 2019-09-23:
2: If its not broken, don't fix it. Serial over USB is more than adequate to handle the amount of data required for klipper. Unless you're having issues with Serial over USB or want to free up a USB port on your pi there's no strong reason to switch.

3: yes

I don't know about 4, but octoprint has a configuration option where you can specify 'long running commands' to prevent octoprint from timing out. Simply add `G1` and `G0` to that list and you should be fine.

karabas2011, 2019-09-23:
> 2: If its not broken, don't fix it. Serial over USB is more than adequate to handle the amount of data required for klipper. Unless you're having issues with Serial over USB or want to free up a USB port on your pi there's no strong reason to switch.

Ok but I cannot set hw serial to work at all. Yes now I am on USB, but I want possibility to switch.

> 
> 3: yes
> 
> I don't know about 4, but octoprint has a configuration option where you can specify 'long running commands' to prevent octoprint from timing out. Simply add `G1` and `G0` to that list and you should be fine.
Yes I was set G1 to "long" on smothieware. But Foosel said it's incorrect and better use firmware with BUSY support.

FHeilmann, 2019-09-23:
2: During the `make menuconfig` step when you're compiling the MCU firmware, you have the option to enable or disable USBSerial. Disable it here and you should be able to talk to your SBase via Serial instead of USB. From a short look at the source code for the lpc176x you will only be able to use UART0 this way though. So if that Serial port is not broken out on your SBase or is used otherwise you won't be able to use direct serial.

3: Another possibility is to simply split your long move into short ones. Klipper and octoprint have a few areas where they don't work together perfectly and you have to use these workarounds.

karabas2011, 2019-09-23:
> 2: During the `make menuconfig` step when you're compiling the MCU firmware, you have the option to enable or disable USBSerial. Disable it here and you should be able to talk to your SBase via [...]
It's the first I had tried. I used hw serial  successfully with smoothieware, then switched to klipper and failed

> 
> 3: Another possibility is to simply split your long move into short ones. Klipper and octoprint have a few areas where they don't work together perfectly and you have to use these workarounds.
Some strange slicer settings required?

closed 2019-09-28 by klipper-gitissuebot