Klipper3d/klipper #1999: How to localize lcd menu?

karabas2011, 2019-09-21:
What files I need to translate to my language to see cyrillic LCD menu?

FHeilmann, 2019-09-23:
your klipper installation has the following file: `klippy/extras/display/menu.cfg`
There you can translate all the lines that start with `name` to localize the LCD menu. Unfortunately klipper does not yet have a localization system so you'll have to do this by hand for your printer.

karabas2011, 2019-09-23:
Thank you very much

karabas2011, 2019-09-24:
But it seems that I need to switch LCD to second code page. LCD2004 has two hardcoded charset pages.  F.e. rrd smart constroller has second japanese page.  Mine one has cyrillic. Marlin switches normally.

closed 2019-09-28 by klipper-gitissuebot

KevinOConnor, 2019-09-28:
It would require code changes to modify the character set on the lcd2004 display.  If someone is interested in doing that, the code is in klippy/extras/display/hd44780.py .

-Kevin