Klipper3d/klipper #2000: no /tmp/printer in octoprint

damik422, 2019-09-21:
Hi, i have a problem regarding setup of klipper on a raspberry pi.
When i'm at the step of adding an additional serial port in octoprint, the port doesn't show up in the roll down menu. Only /dev/ttyUSB0 shows up. What could be the problem here? Is there some common solution for this problem or do i need to provide more info about the system? I know that the port should be visible in /tmp/ but it isn't

Raspberry pi version 3B+
using a fresh install of octoprint

damik422, 2019-09-21:
There is also no klippy.log here, because i can't even connect to the proper port

cosminr86, 2019-09-21:
You must add it manually. 
Carefully read install docs

damik422, 2019-09-22:
> You must add it manually.
> Carefully read install docs

@cosminr86 if you mean by that adding it in additiona serial ports in octoprint then i did it and it still doesn't show up. If you mean something else then I don't think there is anything about it in the installation guide.
[image]
[image]

BlackStump, 2019-09-22:
You have follow the install steps 
https://github.com/KevinOConnor/klipper/blob/master/docs/Installation.md#prepping-an-os-image
and done this part right?
>git clone https://github.com/KevinOConnor/klipper
>./klipper/scripts/install-octopi.sh
if you have there should be a log

closed 2019-09-28 by klipper-gitissuebot