Klipper3d/klipper #2002: Question - General Input pin type

RadicalZephyr, 2019-09-22:
Hello, I'm currently digging through the Klipper source code to try and understand how much work it would take to use Klipper to control my printer.

In particular I have two requirements that don't seem to have a simple way to set up currently in Klipper.  Firstly, my printer has a door with a NO switch in the latching mechanism so that when the door is closed the circuit is closed.  I would like to be able to detect in Klipper when the door opens (and the pin changes state).  It seems like I could maybe write a custom module that sets up that pin as an 'endstop' type, but then it's not totally clear to me how I would use that `MCU_endstop` to register for state change notifications.

Secondly, I have a more complex feeding mechanism that involves two output pins (one PWM, one digital) and one input pin for a sensor.  Again, I would like to be able to write a custom module where I can register to receive callbacks when that sensor pin changes state (binary input).

From the digging I've done through the pin setup code it looks like this would involve adding an `MCU_digital_in` class to `mcu.py` and the corresponding setup code to create pins of that type.  However, I suspect that there may also be changes needed to the microcontroller code to support such a new pin type.

Any advice on what might be needed for this kind of addition would be much appreciated.

I'm also interested in whether you think it would be worth including in Klipper by default. I'd be happy to try and put something together and submit a PR for it.

KevinOConnor, 2019-09-22:
For the input pins, it sounds like you want to look at the gcode_button config section (described briefly in config/example-extras.cfg).  Depending on the action to be taken, though, you may need to implement a new python module - take a look at klippy/extras/buttons.py for registering callbacks on input events (which gcode_button also uses).

If it's a filament sensor, you might want to look at klippy/extras/filament_switch_sensor.py.

In general, we're open to adding new "extras" modules.  I'd need more details on the hardware in order to give more specific feedback though.

Cheers,
-Kevin

RadicalZephyr, 2019-09-23:
Ah! Buttons looks like what I need for right now.  Thanks for the pointer.

I was talking about contributing a 'digital_input' type so that an extras module could just configure a pin like
```python
ppins.setup_pin('digital_input', config.get('sensor_pin'))
```

I don't think the code I'm writing for controlling my extruder makes sense to be a Klipper extra since it's a totally bespoke setup for feeding the extruder using pellets instead of filament.

KevinOConnor, 2019-09-25:
> I was talking about contributing a 'digital_input' type so that an extras module could just configure a pin like

Ah, okay.  The current code uses the buttons.py code to accomplish that.

-Kevin

closed 2019-09-26 by RadicalZephyr