Klipper3d/klipper #2004: [feature request] combine bed_tilt and bed_mesh features

jperrin72, 2019-09-23:
I'm using the bed_mesh feature and find it very useful. I have to check and calibrate my bed on a regular basis. I tried to enable bed tilt on top but I get an error "G-Code move transform already specified".
I understand the bed_tilt code is trying to override the existing bed_mesh instead of updating it.

I wonder if both features could be combined through the following sequence:
1) Klipper initialized with default bed mesh (current behavior) 
2) on BED_TILT_CALIBRATE command bed is probed to create a tilt mesh
3) depending on the bed probe:
  - if tilt>max allowed:  Print job is  canceled to allow manual bed screw calibration by user (SCREWS_TILT_CALCULATE)
  - if tilt<=max_allowed : both bed and tilt meshes are combined together for z correction

Would it be possible to amend the code to do this ?

Thanks in advance

Jerome

KevinOConnor, 2019-09-25:
I'm sure it's possible.  A developer would need to implement and submit this.

-Kevin

jperrin72, 2019-09-26:
I think I could try to code it but might need some help with code snippet that show how to cancel a print job.

Hywelmartin, 2019-09-27:
In Marlins ABL you tilt the mesh according to a 3 point probing at the beginning of each print... so more of an extension of the mesh-bed leveling..

jperrin72, 2019-09-27:
Hi Hywel, Could be however I was thinking to apply both module mesh transformation to tool head coordinates (tilt then mesh) to avoid an heavy change on the bed-mesh code and make testing easier. What do you think?

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

Hywelmartin, 2019-09-27:
@jperrin72  I'm not a coder so I don't know... And I don't use either.. It was just a suggestion..

jperrin72, 2019-09-28:
just created pull request #2018 with a code change that seems to work well for me so far

Arksine, 2019-09-28:
Question:  Instead of using the default mesh, why not probe a new mesh prior to each print?  I fear that stacking bed_tilt and bed_mesh transforms could result in too much adjustment applied to Z depending on the tool's location.

jperrin72, 2019-09-28:
>Question:  Instead of using the default mesh, why not probe a new mesh prior to each print?  I fear that stacking bed_tilt and bed_mesh transforms could result in too much adjustment applied to Z depending on the tool's location.  

@Arksine this takes a long time to probe a new mesh - I’ve been using stored mesh and it works well as long as the bed is correctly leveled. About having too much adjustment on Z can be avoided with the tilt tolerance feature I was thinking about. so fat My first test of the pull requests works well.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

jperrin72, 2019-10-05:
@KevinOConnor Hi Kevin, I updated the commit with a proper sign-off for your review and hopefully commit

KevinOConnor, 2019-12-23:
I'm going to mark this as inactive, as it looks like the conversation has concluded for now.

-Kevin

closed 2019-12-23 by KevinOConnor