Klipper3d/klipper #2006: Re: Clarification needed on Z-offset changes (i.e., baby stepping)

CalielOfSeptem, 2019-09-23:
This is not a bug, but instead a request for clarification.  I generally watch my first layers which is set to a very low speed of maybe 10mm/s, so I can tweak z-offset via baby stepping.

Klipper does not 'act' like a Duet or standard MCU (like a stock ATMega of my Ender 5) in that it seems to wait a hella-long time to apply the change to the z-offset.  If I was to speculate, the z-offset changes between layers.  This makes it difficult to baby step live and see the result... which is a crutch that I would like to have.

So why does Klipper have this delay, and am I just missing a setting that would correct the delay?

mental405, 2019-09-23:
Two reasons. 

The SET_GCODE_OFFSET command requires a parameter ( MOVE=1 ) to be set in order to execute the movement immediately. Otherwise it will wait until the next time and absolute move is executed (ie: The next layer height change).

So, if you are babysitting the first layer (that was not a typo) and adjusting the offset, you have to include MOVE=1 for it to take effect immediately.

Second reason. It still won't take effect immediately. Klipper uses a very tight movement timing and queue system. Many moves are caculated and scheduled ahead of time and inserted into the queue. When you adjust your Z offset, it is inserting that change to the end of the queue. You have  to wait for that move to get executed before you will see a result.

If you have repeatability issues on your Z axis, you may wish to look into the Endstop Phase feature if you happen to use Trinamic drivers.

CalielOfSeptem, 2019-09-23:
> Two reasons.
>
> The SET_GCODE_OFFSET command requires a parameter ( MOVE=1 ) to be set in order to execute the movement immediately. Otherwise it will wait until the next time and absolute move is [...]

Thanks! I will try the options you listed and see what works best.

BlackStump, 2019-09-23:
@CalielOfSeptem 
it is noted in the Config_Changes doc
https://github.com/KevinOConnor/klipper/blob/master/docs/Config_Changes.md

>20190520: The SET_GCODE_OFFSET command has changed; update any g-code macros accordingly. >The command will no longer apply the requested offset to the next G1 command. The old behavior >may be approximated by using the new "MOVE=1" parameter.

bioghost, 2019-09-25:
You can modify your printer.cfg as detailed in this comment in order to override the default z offset menu. I use this and it works pretty good, one thing to be wary of is that the offsets are not applied immediately, and sometimes cause a momentary stutter in the X/Y movement. The amount of time it takes for the offset to be applied seems to relate to the feature being printed. Certain circles or curves will print completely before the Z motor turns. I print things with a rather thick skirt and perform the offset tuning while that is being printed to avoid screwing up the first layer of the print on accident.

https://github.com/KevinOConnor/klipper/issues/1735#issuecomment-502459356

KevinOConnor, 2019-10-08:
I'm going to close this as it appears the original questions were answered.

-Kevin

closed 2019-10-08 by KevinOConnor