Klipper3d/klipper #2011: Compiling klipper.bin with 0 microsecond delay introduces X/Y drift

CalielOfSeptem, 2019-09-25:
I am running an SKR pro with TCM5160s in SPI.  Running the latest klipper build and re-flashed by SKR with the microseconds set to zero.  

All of the prints following that show significant X- drift.  Initial layer starts off OK, but then its quickly visible that some of the 'brim' lines are a bit out-of-whack.  It goes down hill from there.

I re-flashed with the microseconds set to 2 (default) and the drift problem has gone away.  Log file attached.. Did I miss something in my setup?
[klippy.log](https://github.com/KevinOConnor/klipper/files/3649945/klippy.log)
[image]

KevinOConnor, 2019-09-25:
I guess I was wrong - the STM32F407 is too fast to disable the minimum step pulse durations even with Trinamic stepper drivers.

You can still set the delay value to 1 which will roughly double the number of steps one can send to a single stepper motor.  There is no general performance improvement from doing this however.

-Kevin

evandene, 2019-09-27:
Dear Ken,
I'm not that familiar with KLIPPER and setting up new firmware for the SKR Pro v1.1 could you please help me with a few things?

- I ordered the TMC5160 drivers and I like to understand the why and when for SPI mode.
- Where do I need to set the microseconds you are talking about?
- Could you share your config file in order to get my pin numbers of the SKR Pro v1.1 right?
- Can I just copy the .bin file on the sd cart, flash the SKR board and update the config file via shh/Raspberry?
- Another "Dummy" question; I have also the beegtree TFT35 touch screen connected. Does Klipper provide support this screen?

CalielOfSeptem, 2019-09-27:
> Dear Ken,
> I'm not that familiar with KLIPPER and setting up new firmware for the SKR Pro v1.1 could you please help me with a few things?
> 
> * I ordered the TMC5160 drivers and I like to understand the why and when for SPI mode. Anny recommendation for doc's for dummy's?
The TMC5160 drivers do **not** require any changes (soldering wise, pin wise) if you use them in SPI mode.  You simply plug them into the SKR pro in the proper orientation (there is a color difference for the pin/bus connectors and an orientation 'dot' to help with that, see my attached photo).  Unlike other SKR boards (like the 1.3, I believe), the SKR PRO is pre-wired to provide SPI without the annoying and complicated pigtail/daisychain connector that the TMC5160s come with.  So to recap, i) do not alter the TMC5160s by removing pins or soldering/etc, ii) throw out or put aside the daisy chain cable the TMC5160s come with.

To answer your question about "why", the reason I use SPI mode (versus UART) is simply because I wanted to get the board up and running without hassle.   All of @KevinOConnor 's examples for the SKR pro were directed to SPI, and I can't see any negative to running SPI speed/feature wise.  My drivers run near silent and are very smooth.  Further note that the jumpers on the SKR board MUST be corrected from the factory so that they appear as shown in my photograph.  The first pair of pins is for UART and does **not** have a jumper.  The remaining jumpers bridge in the SPI paths.  So be sure to pay close attention to those jumpers.

> * Where do I need to set the microseconds you are talking about?
When you run the '> make menuconfig' step of klipper this is an advanced option.  By default it is set to 2, and you should not change it.
 
> * Could you share your config file in order to get my pin numbers of the SKR Pro v1.1 right?
See attached zip file.  Note, the steppers for X, Y, Z and E are all configured for Ender 5 motors, which is to say NEMA 17 pancakes.  If you have anything other than an Ender 5 and/or have a different extruder motor/setup (like a Volcano, BMG, etc), you will definitely need to adjust the settings for that.

> * Can I just copy the .bin file on the sd cart, flash the SKR board and update the config file via shh/Raspberry?
Once you get the firmware.bin (renamed from klipper.bin) on to the SKR board, it becomes essentially a 'dumb box.'  All settings changes are made at the Pi-level via the printer.cfg which is the pi/home directory (if I recall correctly).  Use a program like notepad++ or other that supports sftp for purposes of remotely editing and uploading the printer.cfg changes.   DO be careful to note that certain commands, such as SAVE_CONFIG, will cause Klipper to update the printer.cfg file by itself (see the bottom most section of my config file).  You can very easily overwrite that section if you accidentally upload an older version that happens to be cached/saved in your text editor.  In the future, I hope Klipper avoids using the printer.cfg in this way as its creates a trap for people who are unaware/new.

> * Another "Dummy" question; I have also the beegtree TFT35 touch screen connected. Does Klipper provide support this screen?
I have one, but did not confirm.  I assume someone has tried, check out the example_extras.cfg (I think that is the name) in the Klipper/Config area.  Note, the TFT35 has an 8-pin connector which will not fit into the SKR pro's 10 pin heads.  My stock Ender 5 LCD screen works well based on my provided config.

See attached config in the zip file.  
[printer.zip](https://github.com/KevinOConnor/klipper/files/3662477/printer.zip)

[image]

evandene, 2019-09-27:
Thanks so much, this info will kick start me when installing Klipper after my drivers have arrived.
The bigtree-tech tft35 v2.0 connector fits as it is a 10 pin connector. I will lookup the samples

evandene, 2019-09-28:
I have uploaded in Thingiverse my interpretation of a 3D printer test bench.
In the upload you can find an accurate SKR-PRO v1.1 Housing front with base and a Power input/monitor back mount or see attached. I was trying to attache here but the system comented that they were to big, so see thingiverse if interested
https://www.thingiverse.com/thing:3879542

[3D PRINTER CONTROLLER TEST BENCH.pdf](https://github.com/KevinOConnor/klipper/files/3665753/3D.PRINTER.CONTROLLER.TEST.BENCH.pdf)

KevinOConnor, 2019-10-19:
I'm going to close this issue as it looks like the original question was answered.

-Kevin

closed 2019-10-19 by KevinOConnor