Klipper3d/klipper #2012: Inverse endstop_pin

andreklug, 2019-09-26:
Hi,

My probe is connected to ar18. It is a LJ18A3-B-Z/AX proximity sensor which worked perfectly in Marlin.

Here's the config:

```
[stepper_z]
step_pin: ar46
dir_pin: ar48
enable_pin: !ar62
step_distance: .0025
endstop_pin: probe:z_virtual_endstop
#endstop_pin: !ar18
position_min: -2
position_endstop: 3
position_max: 200
homing_retract_dist: 1

[probe]
pin: !ar18
x_offset: +31.0
y_offset: -29.
z_offset: -0.79
speed: 10.0

[safe_z_home]
home_xy_position: 100,100
speed: 50.0
z_hop: 5.0
z_hop_speed: 20.0

```

As the sensor is a NO (normal open) sensor, it should be configured as "pin: ^ar18". As you can see, it is configured as "pin: !ar18". Because when I configure it the correct way, the result of "QUERY_ENDSTOPS" is always "TRIGGERED". Only when doing it as above, it differentiates between "OPEN" and "TRIGGERED".
But then of course I am getting the "!! Endstop z still triggered after retract" error.

Is there a way to inverse the output? Or is there a way to get the correct behaviour set?
[klippy.log](https://github.com/KevinOConnor/klipper/files/3656311/klippy.log)

thanks in advance!

BlackStump, 2019-09-26:
have you tried     ^! ar18

andreklug, 2019-09-26:
Yes, without success

andreklug, 2019-09-26:
yeah... I reinstalled from scratch, and with "ar18" it works.

closed 2019-09-26 by andreklug