Klipper3d/klipper #2013: invalid oid type

smark-, 2019-09-26:
I've been working on a fork of Klipper but I stumbled across the invalid oid type error. Could you please provide my with some reference on the oid "workflow" and when such an error gets triggered?

KevinOConnor, 2019-09-29:
The best information we have is the developer documentation: https://www.klipper3d.org/Overview.html#developer-documentation

-Kevin

closed 2019-10-04 by klipper-gitissuebot

cooked, 2019-10-12:
Hi,
after deep diving into the documentation, and reviewing a similar issue #924,  it is still uncler to me if the shutdown of the printer is caused by the invalid oid or, the other way around, the invalid oid is thrown because the printer was already shutdown. 
below the tail of the send and receive queues

[image]

[image]

I don't have enough experience on the MCU code to figure out what is going wrong there, and I don't know how to debug the FW part, so any pointer is appreciated.

Looking forward to your help.

Stef

KevinOConnor, 2019-10-12:
An "invalid oid" error is usually the result of the micro-controller sporadically resetting.  The host doesn't know the mcu did a reset, the host continues sending commands, the micro-controller is no longer in a configured state, and so it reports that error.  The error could be thought of as "a command was sent to the micro-controller for an object id that has not been configured".

-Kevin

cooked, 2019-10-13:
Hi Kevin,
thank you for the insight. Assuming the root cause is the mcu, how do you normally go about debugging mcu code? I can think of using console.py to manually send commands but how to step into the code? Do you use simulavr or a debugger?
Best,
Stef