Klipper3d/klipper #2016: error: Unknown command: endstop_home

C45tr0, 2019-09-28:
Starting serial connect
Loaded MCU 'mcu' 74 commands (v0.7.0-489-g2cff329-20190601_111624-ender3-octopi-1 / gcc: (15:5.4.1+svn241155-1) 5.4.1 20160919 binutils: (2.27-9+9) 2.27)
MCU 'mcu' config: BUS_PINS_i2c1=P0.1,P0.0 STEP_DELAY=2 ADC_MAX=4095 BUS_PINS_ssp0=P0.17,P0.18,P0.15 RESERVE_PINS_USB=P0.30,P0.29,P2.9 CLOCK_FREQ=100000000 MCU=lpc176x STATS_SUMSQ_BASE=256
Protocol error
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 138, in _connect
    cb()
  File "/home/pi/klipper/klippy/mcu.py", line 672, in _connect
    self._check_config()
  File "/home/pi/klipper/klippy/mcu.py", line 613, in _check_config
    self._send_config(None)
  File "/home/pi/klipper/klippy/mcu.py", line 565, in _send_config
    cb()
  File "/home/pi/klipper/klippy/mcu.py", line 170, in _build_config
    " rest_ticks=%u pin_value=%c", cq=cmd_queue)
  File "/home/pi/klipper/klippy/mcu.py", line 716, in lookup_command
    cmd = self._serial.get_msgparser().lookup_command(msgformat)
  File "/home/pi/klipper/klippy/msgproto.py", line 300, in lookup_command
    raise error("Unknown command: %s" % (msgname,))
error: Unknown command: endstop_home
[klippy.log](https://github.com/KevinOConnor/klipper/files/3665067/klippy.log)

[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/3665069/printer.cfg.txt)

FHeilmann, 2019-09-28:
Recompile (make clean && make) and reflash(make flash...) your board. The code on your board is 4 months old and I'm guessing you just updated your Klipper repo on the pi.

C45tr0, 2019-09-29:
Thought I had done that. Didn't check that it took. Thank you!

closed 2019-09-29 by C45tr0